sebanc/brunch #203: [CRITICAL] Recovery Soft-Bricked Desktop

xeu100, 2020-05-27:
Aftermath of #202 after an attempt to re-enter the system.

Events:
1.  System powered off while frozen on splash screen, post #202
2. Using desktop normally, attempted to re-enter the system.
3. On CrOS boot, recovery started saying that there was a firmware issue.
4. After a 5-minute timer in the top left of the display, system booted DIRECTLY to recovery and did the same thing.
5. After 2nd "recovery", the desktop was no longer able to boot at all, no POST, etc. No monitor output. Odd machine noises.

To resolve this issue I had to use jumpers to reset the CMOS. I don't know how, but during some recovery methods, it corrupted the CMOS. My data is okay, I just had to reconfigure my CMOS settings.

This is possibly critical if anyone attempts my patch on an Nvidia system.

sebanc, 2020-05-27:
Indeed, I don't think this is related to your script, however it is weird that you had the chromeos recovery screen. Are you dual booting with windows ? If so, have you disabled fast startup ?

I did not know that at the end of the recovery chromeos would try to update the CMOS. For now, if you see that screen (which is not normal), force shutdown the pc before the end of the countdown.

xeu100, 2020-05-27:
Fast startup was never an option on my device, but yes it is disabled. I was dual-booting with Windows but manually made a separate EFI partition on another drive. This solution worked until it prompted first a partition recovery then it restarted directly (full restart) a firmware recovery. That'll work for now, but I wonder if anything can be done to stop recovery from popping up and just let it error out.

khanhtranngoccva, 2020-05-28:
ChromeOS trying to write to CMOS does cause tons of problems, because we don't use their BIOS. Can you disable all CMOS updates?

xeu100, 2020-05-29:
The BIOS has flashing disabled for security, but beyond that recovery should be disabled from ChromeOS itself.

sebanc, 2020-06-28:
Forgot to update this issue but all CMOS update features are disabled in recent brunch builds.

closed 2020-06-28 by sebanc