sebanc/brunch #207: Fonts

Pumpino, 2020-05-27:
I'm test driving Brunch. I've noticed that the fonts in Chromium appear different to how they do in CloudReady and Arch linux. They're thinner and less sharp, despite the same (default) fonts being used.

Is there any way I can change this?

sebanc, 2020-05-27:
Hi, I have searched the usual places but could not find specific options / flags related to font rendering.

If you still have cloudready, could you post the output of chrome://version from there ?

Pumpino, 2020-05-27:
Chromium	80.0.3987.173 (Official Build) (64-bit)
Revision	0
Platform	80.4.1 (Developer Build - neverware) stable-channel chromeover64
Firmware Version	
JavaScript	V8 8.0.426.30
Flash	20200512.1.999.999 /opt/google/chrome/pepper/libpepflashplayer.so
User Agent	Mozilla/5.0 (X11; CrOS x86_64 12739.118.0) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/80.0.3987.173 Safari/537.36
Command Line	/opt/google/chrome/chrome --use-gl=egl --gpu-sandbox-failures-fatal=no --enable-logging --log-level=1 --use-cras --enable-wayland-server --user-data-dir=/home/chronos --enable-features=SmartDim,Crostini --system-developer-mode --login-profile=user --default-wallpaper-is-oem --aura-legacy-power-button --default-wallpaper-large=/usr/share/chromeos-assets/wallpaper/oem_large.jpg --default-wallpaper-small=/usr/share/chromeos-assets/wallpaper/oem_small.jpg --default-wallpaper-is-oem --arc-build-properties={} --enterprise-enrollment-initial-modulus=15 --enterprise-enrollment-modulus-limit=19 --login-manager --first-exec-after-boot --vmodule=*/search_result_ranker/*=1,*night_light*=1,*/chrome/browser/chromeos/account_manager/*=1,*/chromeos/components/account_manager/*=1,app_list_syncable_service=1,*/chromeos/power/auto_screen_brightness/*=1,*/forced_extensions/installation_tracker*=2,extension_downloader=2,existing_user_controller=2,*/ash/wm/tablet_mode/*=1,enrollment_screen_handler=1,*/browser/chromeos/login/enrollment/*=1,*/ui/ozone/*=1,*/ui/display/manager/chromeos/*=1,update_engine=1,component_updater_service=1 --enable-features=SmartDim,Crostini --ppapi-flash-path=/opt/google/chrome/pepper/libpepflashplayer.so --ppapi-flash-version=20200512.1 --ui-compositor-memory-limit-when-visible-mb=512
Executable Path	/opt/google/chrome/chrome
Profile Path	/home/chronos/u-017a7635b0e1342b546fb00eab938c0d57d96ffc

Pumpino, 2020-05-27:
The fonts in CloudReady look bolder although not as sharp in the screenshots. I think bolder is better.

Brunch:
https://drive.google.com/open?id=1304Yfu6ClkpnmkkOFPS_LPA0LP8aqFaT
https://drive.google.com/open?id=10W1ijmyla1-qH6wxG07Uj8B_8BHD9iov

CloudReady:
https://drive.google.com/open?id=1-s3RLNkKI4X63XDJwpoAX6LfKOVUpacL
https://drive.google.com/open?id=1y_SZjDCpfmfBAl-_747aKhlNVUXvILuO

sebanc, 2020-05-28:
Thanks for the chromium chrome://version log, unfortunately I see no difference which could impact fonts. However, looking at the pictures, it looks more like a display manager scaling issue which could be explained by the fact that chromiumos/arch do not have the same display manager as chromeos.

Could you try to see if changing the resolution slider in display settings improves this ?

Pumpino, 2020-05-28:
Admittedly, I took the screenshots of CloudReady on my desktop PC with a 24" monitor, and the ones of Brunch on my Dell 12.5" laptop, so that probably didn't help. However, the same issue was present when I was running Brunch on the desktop computer.

I tried changing the resolution and also increasing the zoom. Increasing the zoom appeared to widen characters, but at the expense of them being ridiculously big. ;) 

I'll just have to adjust to the change in appearance. Thanks for looking into it (and for developing such a great project).

sebanc, 2020-05-28:
Since this morning when I updated to r83, I am under the impression that something has changed regarding fonts, maybe it's just an impression but you can try updating.

Pumpino, 2020-05-28:
This is the first time I've tried to update, and I think it failed.

Do I need to enable native updates first? If so, I didn't.

When I run sudo chromeos-update -f, do I decompress the archive first, or is it just sudo chromeos-update -f <path to brunch_r83_k4.19_testing_20200528.tar.gz>?

midi1996, 2020-05-29:
yes, you feed the tar.gz file directly, no need to extract it.

Pumpino, 2020-05-29:
That's weird, as that's what I did and it's it still reports I'm running v80. It did spit out a couple of errors messages when I did it. Is it reporting v80 because that's the recovery version? Is there anywhere i can check that I'm running v83?

sebanc, 2020-05-29:
There are 2 different update types (brunch & chromeos):
- You updated brunch when running "sudo chromeos-update -f < path to brunch_r83_k4.19_testing_20200528.tar.gz >"
- To update chromeos, you need to enable native updates and update through the settings menu.

Pumpino, 2020-05-29:
I'm a bit confused with the roles of, and differences between, Brunch, ChromeOS and recovery. I thought updating Brunch was updating ChromeOS. If Brunch isn't ChromeOS, what is it?

sebanc, 2020-05-29:
ChromeOS is the OS built by google and present in chromebook recovery images.

Brunch is all the elements that needs to be changed/replaced so that a chromebook recovery image can boot on a standard pc (linux kernel, firmware files...)

Pumpino, 2020-05-29:
So when you release an update, you're basically updating the elements needed to make a new version of ChromeOS work on a PC? And enabling native updates is basically updating the recovery image that I used when I first installed everything?

So I add "options=enable_updates" to the kernel command line in grub.cfg on partition 12 (if Google has shown me a correct thread)? And I don't need to change the channel from stable?

Once it's updated, should I disable nativ eupdates to prevent a future updating installing before you've updated Brunch?

Pumpino, 2020-05-29:
Where is grub.cfg?! There's no /boot!

sebanc, 2020-05-29:
> So when you release an update, you're basically updating the elements needed to make a new version of ChromeOS work on a PC? And enabling native updates is basically updating the recovery image that I used when I first installed everything?

Correct

> So I add "options=enable_updates" to the kernel command line in grub.cfg on partition 12 (if Google has shown me a correct thread)? And I don't need to change the channel from stable?

Correct

> Once it's updated, should I disable nativ eupdates to prevent a future updating installing before you've updated Brunch?

Not necessary, It should not update unless you request it.

> Where is grub.cfg?! There's no /boot!

If you are in single boot, just run "sudo edit-grub-config" and add "options=enable_updates" after "cros_debug".

Pumpino, 2020-05-29:
That worked perfectly. Thanks. 

I'm guessing the correct procedure is to always wait for a Brunch update and to apply that before manually checking for updates, since updating ChromeOS prior to a Brunch update becoming available might hose the system?

The fonts look the same to me, but I'll adjust. :)

One last question (it's probably not worth starting a new thread about). I have gFTP pinned to the shelf, but the icon becomes invisible after every restart. There's a space left for it in the shelf, and if I click on where the icon was, it launches gFTP. This doesn't occur in CloudReady. Any ideas?

Pumpino, 2020-06-10:
Just as a follow up to this. I installed Font Rendering Enhancer from https://chrome.google.com/webstore/detail/font-rendering-enhancer/hmbmmdjlcdediglgfcdkhinjdelkiock?hl=en, and set it to 20. This has improved the appearance of fonts for me.