sebanc/brunch #209: accelerometer working but DELAY

siawzen, 2020-05-27:
hello anyone can help my device HID sensor collection VEN_8087&PID_0A04&REV_0001 is WORKING but delay about 1 second how to solve this issue rammus 8.1 and brunch 8.1 thanks

PixelOFC, 2020-05-27:
how u got rammus 81

PixelOFC, 2020-05-27:
do you really need real time accelerometer?

siawzen, 2020-05-28:
How to do that?

sebanc, 2020-05-28:
Could you precise your use case ? In which app are you experiencing this latency ?

siawzen, 2020-05-29:
In cpuz accelerometer value change but delay, i try another app like level meter & labyrinth game is delay but it's work my device acer w700 tablet, thanks for your support

sebanc, 2020-05-29:
I tried to reproduce it but the accelerometer value change is instant for me, the issue might be with your accelerometer. Have you checked if it is not the same in windows/linux ?

sebanc, 2020-05-29:
It's not exactly the same device but indeed try the dual boot method, even if it's just to test, the device speed is most likely the issue.

siawzen, 2020-06-01:
Please give me tutorial step by step to make dual boot using linux mint or windows, thanks

closed 2020-06-26 by sebanc