sebanc/brunch #213: Device compatibility ASUS Transformer Book T200TA

Ebbot9876, 2020-05-28:
At last I could replace Windows with something useful on this detachable tablet.
Brunch r81 k4.19 stable. Rammus 83.0.4103.50
Working out of the box:
- Android apps
- Linux
- Bluetooth (only tested speaker)
- WiFi
- Speakers
- Trackpad
- Touchscreen

Not working:
- Microphone
- Automatic Screen rotation
- Camera

Other issues:
- Had to install rEFInd boot manager to boot from mmc. Internal UEFI system did not find 
  ChromeOS UEFI. USB boot works fine.
- Sometime freeze, have to hard reboot.

sebanc, 2020-05-28:
Hi, could you attach a dmesg to this issue ? (run "dmesg > ~/Downloads/dmesg.txt" and attach dmesg.txt from your downloads folder)

Ebbot9876, 2020-05-28:
[dmesg.txt](https://github.com/sebanc/brunch/files/4694765/dmesg.txt)

Ebbot9876, 2020-05-28:
Maybe lsmod too?
[lsmod.txt](https://github.com/sebanc/brunch/files/4694771/lsmod.txt)

sebanc, 2020-05-28:
Thanks for the dmesg and the lsmod.

Unfortunately, regarding screen rotation, your device does not use the common intel ISH sensors which are the only ones that have been ported to chromeos (I would need an lsmod from a linux distro to tell you exactly which sensors you have). Unless another dev with the same hardware ports the sensor driver, there is not much I can do.

Regarding camera also, chromeos only supports usb cameras which is not the case here.

For microphone, that's a common issue that you can follow here: #56 and that I have not been able to fix yet.

For the freezes, I think that adding "intel_idle.max_cstate=1" or "intel_idle.max_cstate=0" to the kernel command line should improve things.

Ebbot9876, 2020-05-28:
Screen rotation I beleive it is mpu6050, however I use ChromeRotate app!
I attached a lsmod from Endless OS! [lsmodEndless.txt](https://github.com/sebanc/brunch/files/4695362/lsmodEndless.txt)
I really don´t need camera.
I will follow the microphone issue!
By the way i´ll try out Brunch r83 testing right now!
Thanks for the opportunity try Chrome OS on PC hardware.
Maybe I spend some money and buy the Lenovo Ideapad Duet Chromebook later!

closed 2020-05-28 by Ebbot9876