sebanc/brunch #214: Intel AX200 Wi-Fi chip with Wi-Fi 6 support?

EldarAron, 2020-05-28:
After a few days of running this on my MS Surface Go 2 in dual bootable mode I'm very pleased. Do easy to backup in Windows. Just copy the chromeos.img. Experiment. Delete and replace with backup if something goes wrong.

Seem to use a bit more power, but that's expected of something that's not 100% configured to run on the specific machine.

But over to something else, the Intel AX200 WiFi chip with WiFi 6 support. Any idea of when it will be supported. I see that Intel launched this chip April of last year. 

For now I use the iwlwifi_backport option.

sebanc, 2020-05-29:
Hi,

According to the intel website, their linux driver (iwlwifi_backport) which is included with brunch supports Wifi 6 (802.11ax).

However, the 802.11ax implementation appears not to be complete in chromeos, you can follow the implementation progress in the below issues:
https://bugs.chromium.org/p/chromium/issues/list?q=802.11ax&can=2

EldarAron, 2020-05-30:
Thanks for your feedback.

closed 2020-05-30 by EldarAron