sebanc/brunch #215: Flickering screen

geofrey89, 2020-05-28:
Thanks again @sebanc . For the r83 testing update....
I have today updated to that release. I have noticed that the screen starts flickering after a few seconds when the chrome boot screen appears...
Everything else is working ok so far..

sebanc, 2020-05-29:
Does the flickering occur just a few second during boot or is it permanent ?

geofrey89, 2020-05-29:
@sebanc ..just few seconds like 4 or 5 seconds ....

geofrey89, 2020-05-29:
[image]

geofrey89, 2020-06-01:
@sebanc . I found a solution.. I added no_ui_delay..to grub ....as per your release notes. ..

geofrey89, 2020-06-01:
@sebanc . I found a solution.. I added no_ui_del

closed 2020-06-01 by geofrey89