sebanc/brunch #216: RTL8723DE - Can't connect Wi-Fi after disconnect or wake

princepjamie06832, 2020-05-28:
After I updated to Brunch v83 and then add "options=rtl8723de", but I can't connect to network on internal Wi-Fi. I have external Wi-Fi USB working okay.

sebanc, 2020-05-29:
Hi,

What's your laptop model ?

Also the is it that you cannot connect at all or just after disconnect or wake ? (I am confused by the title of the issue)

princepjamie06832, 2020-05-29:
Currently, I have HP Laptop 15-bs115dx (Core i5-8250U).

sebanc, 2020-05-29:
Also is it that you cannot connect at all or just after disconnect or wake ? (I am confused by the title of the issue)

sebanc, 2020-05-29:
Could you attach a dmesg to this issue ? (run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and attach dmesg.txt from your downloads folder)

princepjamie06832, 2020-05-29:
Here's my attachment at my Google Drive: [dmesg.txt](https://drive.google.com/file/d/1DwJMdPmPjfQ4OjkPe7BAhTkvTCAt5FIl/view?usp=sharing)

sebanc, 2020-05-29:
Just to be sure the issue lies within the driver, could you post the output of "sudo rfkill list all" ?

princepjamie06832, 2020-05-30:
When I reconnect or change Wi-Fi network, it refuses to connect Wi-Fi unless restart computer. After wake, sometimes connect to Wi-Fi or not. I will try not to reconnect or change network and keep connected as long as I can. Sleep is working okay. 

Here's I tested

Connected Wi-Fi after startup: OK
Connected Wi-Fi after wake from sleep: Sometimes OK or not
Stay Connected as long as can: OK
Reconnect or Change Wi-Fi Network: Failed

vladimirkosolapov, 2020-06-06:
Same issue with RTL8723BS. ✋🏻

princepjamie06832, 2020-06-18:
I've moved to new computer. This issue is closed.

closed 2020-06-18 by princepjamie06832