sebanc/brunch #218: [Surface Go 2] Issues with waking up from sleep

petertriho, 2020-05-29:
After updating to r83 + Brunch r83 k4.19 testing 20200528, it seems like my SG2 fails to wake up from sleep

petertriho, 2020-05-29:
Seems like the issue only occurs if it sleeps from closing the type cover. Using power button doesn't cause this issue (with suspend_s3 it reboots, without suspend_s3 it shows a black screen)

Reverting back to r83 + Brunch r81 k4.19 testing 20200526 seems to not have the issue

sebanc, 2020-05-29:
Do you know if it was working with ChromeOS r81 + Brunch r81 k4.19 testing 20200526 ?

petertriho, 2020-05-29:
Yep, it was working fine on chrome os r81 + Brunch r81 k4.19 testing 20200526

sebanc, 2020-05-29:
Could you try to run "echo 0 | sudo tee /usr/share/power_manager/board_specific/use_cras" in crosh shell, then reboot and see if it helps with suspend ?

petertriho, 2020-05-29:
That seems to work

sebanc, 2020-05-29:
Well, I already had several suspend issues and apparently r83 is making it worse, I will try to build r84 kernel to see if it helps.

petertriho, 2020-05-30:
I tried changing to nocturne (r83) and still get the same issues. Reverted back to nocturne (r80) and everything is back to normal

petertriho, 2020-05-30:
I've tried reverting back to r80 (but switched to nocturne) + r83 (nocturne) with Brunch r83 k4.19 testing 20200528 + the brunch release from https://github.com/sebanc/brunch-testing/releases and still had the issues waking up so it seems like it must have been some of the changes made after Brunch r81 k4.19 stable 20200510 is causing the issues. 

I've reverted back to Brunch r81 k4.19 stable 20200510 + r80 (nocturne) and it's back to normal.

sebanc, 2020-05-30:
Thanks for those notes, it's very helpful :) Not sure if you are using crostini or an sdcard but if you use them, could you try suspend after disabling crostini and suspend without sdcard ?

petertriho, 2020-05-30:
Crostini has always been disabled

Suspend seems to be working now that I removed the sdcard

petertriho, 2020-05-30:
Nevermind, spoke to soon. Still does not wake up :(

sebanc, 2020-06-01:
Could you try the "brunch-testing" I posted yesterday ?

petertriho, 2020-06-01:
Just tried Brunch r81 k4.19 unstable 20200531 with ChromeOS V83. Seems like the first sleep/wake works fine but subsquent tries, it remains asleep (keyboard lights up but black screen, it sometimes wakes up but then screens turns off again)

sebanc, 2020-06-01:
Well, I don't know if it will change something but try to go back to rammus for testing suspend. Nocturne image has an ipu3 camera, the camera service is running in a loop in the background which is know to increase cpu consumption and might create suspend issues.

petertriho, 2020-06-01:
Should I test rammus v80 or v83?

petertriho, 2020-06-01:
Switched to rammus v80 - still have the issue 

Updating to rammus v83 now to see

sebanc, 2020-06-01:
Yes, as you prefer rammus r80 or r83 should both work. There should be no need to powerwash

sebanc, 2020-06-01:
Also, could you post the output of:
udevadm info /sys/bus/usb/devices/*

petertriho, 2020-06-01:
```
P: /devices/pci0000:00/0000:00:14.0/usb1/1-0:1.0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-0:1.0
E: DEVTYPE=usb_interface
E: DRIVER=hub
E: ID_USB_CLASS_FROM_DATABASE=Hub
E: ID_USB_PROTOCOL_FROM_DATABASE=Single TT
E: INTERFACE=9/0/0
E: MODALIAS=usb:v1D6Bp0002d0419dc09dsc00dp01ic09isc00ip00in00
E: PRODUCT=1d6b/2/419
E: SUBSYSTEM=usb
E: TYPE=9/0/1
E: UDISKS_PRESENTATION_HIDE=0
E: USEC_INITIALIZED=48605024
```

sebanc, 2020-06-01:
What's the output of `dmesg | grep -i "type Cover"` ?

petertriho, 2020-06-01:
Empty for "Type Cover"
But for `dmesg | grep -i "Surface Keyboard"`

```
[    1.679744] usb 1-7: Product: Surface Keyboard
[    1.685615] input: Microsoft Surface Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/0003:045E:09B5.0001/input/input3
[    1.739236] hid-generic 0003:045E:09B5.0001: input,hidraw0: USB HID v1.11 Keyboard [Microsoft Surface Keyboard] on usb-0000:00:14.0-7/input0
[    1.741742] input: Microsoft Surface Keyboard Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.1/0003:045E:09B5.0002/input/input4
[    1.795194] input: Microsoft Surface Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.1/0003:045E:09B5.0002/input/input6
[    1.797302] hid-generic 0003:045E:09B5.0002: input,hiddev0,hidraw1: USB HID v1.11 Device [Microsoft Surface Keyboard] on usb-0000:00:14.0-7/input1
[    1.799795] hid-generic 0003:045E:09B5.0003: hiddev1,hidraw2: USB HID v1.11 Device [Microsoft Surface Keyboard] on usb-0000:00:14.0-7/input2
[    1.804803] input: Microsoft Surface Keyboard Touchpad as /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.3/0003:045E:09B5.0004/input/input8
[    1.807284] hid-multitouch 0003:045E:09B5.0004: input,hiddev2,hidraw3: USB HID v1.11 Mouse [Microsoft Surface Keyboard] on usb-0000:00:14.0-7/input3
```

sebanc, 2020-06-01:
could you post the output of "udevadm info /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7" ?

petertriho, 2020-06-01:
Getting unknown device

sebanc, 2020-06-01:
Could you replace the content of /lib/udev/rules.d/99-auto_suspend.rules with:
```
ENV{ID_MODEL}=="Surface_Type_Cover", ENV{ID_MODEL_ID}=="096f", TEST=="power/control", ATTR{power/control}="auto"
ENV{ID_MODEL}=="Surface_Keyboard", ENV{ID_MODEL_ID}=="09b5", TEST=="power/control", ATTR{power/control}="auto"
```

Then reboot and test suspend again.

petertriho, 2020-06-01:
Seems to working now! (fingers cross). Thanks for the help

sebanc, 2020-06-01:
try "options=suspend_s3", it should be more stable and have better energy saving

petertriho, 2020-06-01:
suspend_s3 seems to cause even more issues. Trying to wake up causes it to just show the windows logo and get stuck there

petertriho, 2020-06-01:
I must have not rebooted after the "Surface_Keyboard" changes before, wakes up like 8/10 times now

petertriho, 2020-06-02:
Have been testing with:
- `ENV{ID_MODEL}=="Surface_Keyboard"` change
- Brunch r83 k4.19 unstable 20200601
- `options=iwlwifi_backport,force_tablet_mode,advanced_als`

 Issues with sleep appear to be gone bt appear again when I'm charging the SG2

sebanc, 2020-06-03:
Not sure what the issue with power might be related to. If it happens on some occasions that you can wakeup the device from suspending while connected to power, could you post a dmesg taken after wakeup ?

petertriho, 2020-06-03:
I just tried to wake up it up from sleep with power connected and it seems to be working normally now.

If I see the issue again, I'll try to record the dmesg. I'll do more testing as I made some changes over the last day (updated to windows 10 2004, updated grub, removed retroarch android app). Removed retroarch android app because when I managed to wake it up the device would be very laggy and I would see "android.process.media" stopped working and retroarch is an app I installed in a close timeframe from when I started having wake issues)

petertriho, 2020-06-03:
Can't manage to pull the dmesg logs, I managed to run the "dmesg > dmesg.txt" but then I reboot and the files is not there. I noticed the WiFi was showing "connecting", apps (both chromeos and android apps) would not open

petertriho, 2020-06-04:
Cannot pull dmesg, every time I try it says the storage is read-only before the whole device freezes

I pulled `/var/log/messages`. Apparently it's dmesg history(?), not sure what I'm looking at but it does seem like it's related to being unable to unmount the SD Card. I'll try removing the SD Card again and seeing if the issue still persists.

```
2020-06-04T13:40:46.025593+10:00 WARNING bluetoothd[2462]: Couldn't open /sys/class/bluetooth/hci0/prepare_for_suspend: No such file or directory
2020-06-04T13:40:46.030230+10:00 ERR cros-disks[2144]: Cannot unmount '/media/removable/SD Card' with flags 0: Device or resource busy (16)
2020-06-04T13:40:46.030348+10:00 WARNING cros-disks[2144]: Mount point '/media/removable/SD Card' is busy, using force unmount
2020-06-04T13:40:46.361214+10:00 NOTICE powerd_suspend[17235]: Going to suspend-to-RAM state: args=--suspend_to_idle --wakeup_count=33
2020-06-04T13:40:46.403271+10:00 NOTICE powerd_suspend[17235]: Available order 3 pages: 8833
2020-06-04T13:40:46.406729+10:00 NOTICE powerd_suspend[17235]: Finalizing suspend
2020-06-04T13:40:46.412742+10:00 INFO kernel: [ 8465.613883] PM: suspend entry (s2idle)
2020-06-04T13:48:22.203618+10:00 NOTICE kernel: [    0.000000] Linux version 4.19.122-brunch-sebanc (sebanc@tablet) (gcc version 8.3.0 (Ubuntu 8.3.0-6ubuntu1)) #1 SMP PREEMPT Mon Jun 1 09:11:53 CEST 2020
```

[messages.txt](https://github.com/sebanc/brunch/files/4727458/messages.txt)

petertriho, 2020-06-04:
Looks like it's an issue with mounting sd cards to andorid apps (Settings > Device > Storage Management > External storage preferences). With external storage perferences disabled or sd card physically removed from device, I haven't encountered the sleep issues yet.

sebanc, 2020-06-04:
It is a very interesting finding. Actually the "External storage management menu" is an unfinalised feature that I activated by default due to an old issue I had.

Could you try to see if you still have the issue after removing the "--enable-features=ArcUsbStorageUI" line from /etc/chrome_dev.conf ? (the menu will not be present anymore but your external storage will be mounted in android by default)

petertriho, 2020-06-04:
I removed the line but menu is still present

petertriho, 2020-06-05:
With the SD Card disabled (via External storage preferences), the SD Card is still avaliable through some apps like solid explorer but disabled in pocketcast (seems like they are mounted at different places)

(formatted my sd card to be named SD_Card)
Disabled -> available through `/storage/removable_SD_Card`
Enabled -> also available through `/media/removable/SD_Card`

It appears that it doesn't matter whether or not the option is disabled/enabled, if I access the sd card via any of these directories, it will eventually cause sleep issues.

sebanc, 2020-06-05:
Ok, I must not have noticed that this feature was now live and enabled by default. Now that you mention it, I remember that someone had to reformat his sdcard to have suspend working correctly, did you notice an improvement after formatting ?

petertriho, 2020-06-06:
Physically removed my sd card yesterday and haven't had a sleep issue since. Even after reformatting the card (to exFAT), I still occasionally had sleep issues.

petertriho, 2020-06-28:
Been testing with the latest brunch-testing release, have had an issue with sleep (even with sd card connected)

closed 2020-06-28 by petertriho