sebanc/brunch #219: force tablet mode doesn't work

widifirmaan, 2020-05-29:
option force_tablet_mode does not work, how to fix it/

sebanc, 2020-05-29:
Hi, do you use dual boot or single boot ? could you post your grub config ?

widifirmaan, 2020-06-01:
heres my grub config...
menuentry "ChromeOS" {
	img_part=/dev/sda3
	img_path=/chromos.img
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	linux (loop,gpt7)/kernel boot=local noresume noswap disablevmx=off \
		cros_secure cros_debug options=force_tablet_mode,enable_updates loop.max_part=16 img_part=$img_part img_path=$img_path
	initrd (loop,gpt7)/initramfs.img
}

tablet mode cant enable after i use sysfs_tablet_mode,after I delete it, it's the same

sebanc, 2020-06-01:
What's your laptop model ?

sebanc, 2020-06-01:
Also, could you post a screenshot of your desktop with force_tablet_mode enabled ?

closed 2020-06-26 by sebanc