sebanc/brunch #222: Macbook Air Wireless

memphisraynz, 2020-05-29:
I just upgraded brunch from one of the 81 releases to the latest 83 release.
i was using options=broadcom_wl and it did work, now after upgrading brunch I cant seem to get it working again.
Any assistance you can provided would be great.
I have loved using this. Also noticed that the keyboard backlight now works, thanks.

sebanc, 2020-05-29:
Hi,

What's the output of "lsmod | grep broadcom_wl" ?
Also what is the output of "sudo rmmod broadcom_wl && sudo modprobe broadcom_wl" ?

memphisraynz, 2020-05-29:
the first command returned no output
the second returned - rmmod: ERROR: Module broadcom_wl is not currently loaded

sebanc, 2020-05-29:
Sorry my mistake, could you reboot and provide the output of:
- "lsmod | grep wl"
- "sudo rmmod wl && sudo insmod /lib/modules/broadcom_wl.ko"

memphisraynz, 2020-05-29:
much the same results,
nothing from the first command and rmmod: ERROR: Module wl is not currently loaded from the second

sebanc, 2020-05-29:
Are you in dual boot or single boot ? Could you post your grub config ?

memphisraynz, 2020-05-29:
Dual booting with MacOS

set timeout=2

menuentry "ChromeOS" {
  regexp --set disk "(^.+)(,gpt)" $root
  linux ($disk,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug options=broadcom_wl,enable_updates
  initrd ($disk,7)/initramfs.img
}

sebanc, 2020-05-29:
If you are dual booting from an img the config should the grub config to modify should be the one provided by the chromeos-install.sh script. Could you explain how you dual boot macos ?

memphisraynz, 2020-05-29:
Originally from MacOS created a new partition, installed Linux Mint on that then followed a guide from these guys - https://www.reddit.com/r/chromeosfans/comments/ga8m06/how_to_install_windows_10_chrome_os_official/

sebanc, 2020-05-29:
I don't know this guide well but I would say that the grub config you have to modify with options=broadcom_wl is the one from linux mint, it should be something like:
```
menuentry "ChromeOS (boot from disk image)" {
	img_part=/dev/xxx
	img_path=/chromeos.img
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
		cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
	initrd (loop,gpt7)/initramfs.img
}
```

memphisraynz, 2020-05-29:
so you're saying modifying the grub using sudo edit-grub-config isnt actually doing anything?
Is there a way I can manually do what the options=broadcom_wl is doing?

sebanc, 2020-05-29:
Yes "sudo edit-grub-config" is only for single boot, for dual boot you have to input the grub config issued by the chromeos-install.sh script somewhere in the linux install grub and that's where you need to add the options.

There is no easy way to do that manually and also it's important that you know where your grub config is.

memphisraynz, 2020-05-29:
Thank you very much, i'll look to get back into Linux and edit grub

memphisraynz, 2020-05-30:
Adding the options to the Linux grub resolve the issue.
Thank you very much

closed 2020-05-30 by memphisraynz