sebanc/brunch #225: Restart loop using USB flash drive on Toshiba NB505 Intel Atom

pxsepulv, 2020-05-29:
Hi
Thanks so much for your codes. I want to test Chrome OS on an old toshiba netbook and I followed all your instructions for legacy bios and a 16GB USB (I want to test the OS before installing). Everything worked until my laptop started a restart loop. I don't know what to do now. I'll really appreciate your help.

Tech spec NB505 Toshiba
1 GB ram
Windows 10
Intel Atom (2011?)
BIOS Legacy (no UEFI available)

In one part of the installation process says 
[sdc] write cache: disabled, read cahe: enabled, doesn't support DPO or FUA
I can't read the rest of the screen because it goes too fast and then restarts and boots from the beginning.

sebanc, 2020-05-29:
Hi,

Which recovery image did you use ?

pxsepulv, 2020-05-29:
I used
chromeos_12739.111.0_samus_recovery_stable-channel_mp-v3.bin.zip

sebanc, 2020-05-29:
Have you unzipped the recovery before running chromeos-install.sh ?

pxsepulv, 2020-05-29:
Yes

sebanc, 2020-05-29:
Ok I will need a short video of the boot process, i am not sure what could be the issue ?

pxsepulv, 2020-05-30:
I tried creating an ISO image too and it was pretty much the same. Here's the link for the video.
[https://youtu.be/v-LIQZwM-30](url)

sebanc, 2020-05-30:
Thanks for the video. It's the first time I see this cpu, I thought at first that it should be compatible but it apparently does not support cpu features/instructions which are used in samus recovery image ("SIGILL si_code=ILL_ILLOPN" message).

Unfortunately as samus is the most compatible recovery image currently, your device will not work with brunch.

If you are looking for android apps support, FydeOS or "PrimeOS Classic" might be good alternatives.

pxsepulv, 2020-05-30:
Thank you so much for your help. I'll try those OS. And thanks again for your codes. I could try chrome OS in another laptop and works perfectly.

pxsepulv, 2020-05-31:
I've been thinking about this issue and I'd like to ask you if you can give some link where I can learn to adapt Chrome OS for devices like mine. I know there are some chromebooks with this CPU and this is a great challenge for me at this moment. I'd really appreciate your help.

sebanc, 2020-06-01:
The issue here is you cannot adapt chromeos images as there is no current chromeos image compatible with your cpu.

The only solution would be to build chromiumos from source without those cpu instructions and add the android apps layer to it (currently only FydeOS succeeded in doing this, they might be able to help).

pxsepulv, 2020-06-01:
I understand. Thank you anyway.

closed 2020-06-02 by sebanc