sebanc/brunch #227: ChromeOS launches UI before all components required for Android are loaded

khanhtranngoccva, 2020-05-30:
Normally, in Chromebooks, these components takes nearly no time to load at all, so one can log in right after the UI appears and experience no problems with the Android container. But custom computers running Chrome take up to 1 minute to fully load them. The last component integral for Android to be loaded before first sign-in is "capabilities", and 30 secs after the UI appears, it spawns a log line stating that it is loaded and "using 32-bit capabilities".

Skawtnyc, 2020-05-30:
Yes, this is normal behavior for Chrome OS. It makes sense since Android configuration is tied to the account logging in.

khanhtranngoccva, 2020-05-31:
[image]
It takes an abnormal amount of 30-80 seconds since the blue step started for the red step to finish. If I try to sign in earlier, Android container won't work. The problem here is how to make the red step finish before the blue one starts.

Skawtnyc, 2020-06-01:
I've been working on a project to automatically set the DPI upon bootup to fix the Android scaling that the devs broke and refused to fix. Turns out that by default any DPI changes would be saved past reboot, but they inserted a line into an init file that resets the DPI to 240 (regardless of device resolution). This init file (and others) only trigger AFTER you've logged in.

The problem for me was that I was having the damnedest time trying to figure out when my init script ran. We only have access to the ChromeOS init scripts in dev mode after enabling read/write. And they all execute before the init scripts in the Android container. There is no way around this, either. The solution was to add a sleep command at the beginning of my script, which basically made it wait long enough for the Android init scripts to finish before executing. My Surface Go needs 10 seconds. My Lenovo Duet needs 20.

khanhtranngoccva, 2020-06-01:
Then how can I do it please? I've figured out the cause, but I haven't worked out any solutions.

Skawtnyc, 2020-06-01:
What are you trying to accomplish? My script is specifically for addressing Android scaling.

sebanc, 2020-06-26:
Duplicate of #164

closed 2020-06-26 by sebanc