sebanc/brunch #229: How do I update my Chrome OS to V83?

leosel, 2020-05-30:
How do I update my Chrome OS to V83 can you please help me

petertriho, 2020-05-30:
Add enable_updates to options https://github.com/sebanc/brunch#framework-options

OR

Wait for 83 recovery image (80 is the latest avaliable for most devices) https://github.com/sebanc/brunch#framework-options

closed 2020-05-30 by leosel

au79912, 2020-05-30:
> Add enable_updates to options https://github.com/sebanc/brunch#framework-options then update through settings
> 
> OR
> 
> Wait for 83 recovery image (80 is the latest avaliable for most devices) https://github.com/sebanc/brunch#framework-options

sorry i am new to all this can you specify how you can add the command to the kernel?

petertriho, 2020-06-01:
I'm assuming you have a basic set up going (just single boot)

If so:
- Ctrl + Alt + T in the chrome browser
- `shell`
- `sudo edit-grub-config`
- Add `options=enable_updates` next to `cros_debug` (space after `cros_debug`)