sebanc/brunch #230: ChromeOS was not found on any device

Khang5687, 2020-05-30:
There's a problem I had to deal recently with Chrome OS so I decided to reinstall Chrome OS via Linux. I ran ```
sudo bash chromeos_image_tester.sh <path to chrome_recovery_image.bin>
``` (the script was written by you to test if the recovery image is compatible with the system) and the output was "ok". I downloaded the newest brunch release r83 and used rammus r80, everything went fine...until I got the error ChromeOS was not found on any device...Can you help me out of this situation?

Khang5687, 2020-05-30:
[image]

khanhtranngoccva, 2020-05-30:
Try to enable USB 3.0 controller by default instead of Smart Auto and change SATA mode to IDE.

sebanc, 2020-05-30:
Hi,

I had forgotten about the "chromeos_image_tester.sh" gist. It is an unfinished work which for now can only tell if your system is compatible with eve recovery image (when run against this image) but it does not work for other kinds of cpu instructions. What is your cpu model ?

The "chromeos is not found" message does not seem related to cpu instructions in that case, are you trying to boot from usb or hdd or dual booting ?

Khang5687, 2020-05-30:
I'm dual booting. And my CPU is Intel Core i3 8145U

sebanc, 2020-05-30:
rammus is perfect for this cpu. your issue is with the grub config you are using which does not seem to be a dual boot one or is incorrect.

Could you post your grub config ?

Khang5687, 2020-05-30:
Here's my 40_custom file
``#!/bin/sh
exec tail -n +3 $0
# This file provides an easy way to add custom menu entries.  Simply type the
# menu entries you want to add after this comment.  Be careful not to change
# the 'exec tail' line above.
menuentry "Chrome OS" --class android-x86{
	img_part=/dev/sda7
	img_path=/chromos.img
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
		cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
	initrd (loop,gpt7)/initramfs.img	
}``

Khang5687, 2020-05-30:
> #!/bin/sh
exec tail -n +3 $0
# This file provides an easy way to add custom menu entries.  Simply type the
# menu entries you want to add after this comment.  Be careful not to change
# the 'exec tail' line above.
menuentry "Chrome OS" --class android-x86{
	img_part=/dev/sda7
	img_path=/chromos.img
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
		cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
	initrd (loop,gpt7)/initramfs.img	
}

Khang5687, 2020-05-30:
#!/bin/sh
exec tail -n +3 $0
# This file provides an easy way to add custom menu entries.  Simply type the
# menu entries you want to add after this comment.  Be careful not to change
# the 'exec tail' line above.
menuentry "Chrome OS" --class android-x86{
	img_part=/dev/sda7
	img_path=/chromos.img
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
		cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
	initrd (loop,gpt7)/initramfs.img	
}

sebanc, 2020-05-30:
I think you need to add " \" after disablevmx=off.

Khang5687, 2020-05-30:
I don't know how to comment my full grub config. Lemme add an image real quick

Khang5687, 2020-05-30:
[image]

sebanc, 2020-05-30:
This config looks fine, have you run "sudo update-grub" in your linux distro ?

Khang5687, 2020-05-30:
Resolved. Magically, I have 2 drives, the main one is the one I install chromeos on. When I install, i chose it to be installed on sda7 partition, but the drives magically swap from sda to sdb and therefore results in an error which described above...I ran gparted and changed sda in the grub config file to sdb and it worked! Btw thanks for your replies. Love you

closed 2020-05-30 by Khang5687

vanimarsingh, 2021-09-07:
There's the same problem I am dealing with pls help , I tried to install chrome os using your method but it bootloops ,but when I open debug mode it shows (chrome os was not found ,falling back to shell )somthing like this I tried both samus and Tammy's image to flash but the same problem occurs I have a Pentium e5300 processor pls helppppp😟😟😟

vanimarsingh, 2021-09-07:
Waiting for your reply

vanimarsingh, 2021-09-07:
Still waiting

gustavofalcao1, 2021-09-25:
Hi. I managed to solve the problem of the disk not being recognized as a bootable disk. Using this guide:
https://itsfoss.com/no-bootable-device-found-ubuntu/

At the end of the boot creation described in the guide. I disabled boot security so the kernel could boot