sebanc/brunch #231: options=enable_updates not working

PixelOFC, 2020-05-30:
I tried using sudo edit-grub-config and tried changing oon my grub entry (on linux).

khanhtranngoccva, 2020-05-30:
Mount the efi partition manually, that will be faster.

PixelOFC, 2020-05-30:
bruh, i used crosh and nano

khanhtranngoccva, 2020-05-30:
How about linux mint?

PixelOFC, 2020-05-30:
i use ubuntu

khanhtranngoccva, 2020-05-30:
Any linux and GUI text editors would work, but don't use Windows to edit bootable stuffs.

khanhtranngoccva, 2020-05-30:
Or I can send you my config.
[grub.txt](https://github.com/sebanc/brunch/files/4705695/grub.txt)

PixelOFC, 2020-05-30:
this is from the linux grub or from the cros?

khanhtranngoccva, 2020-05-30:
This is CrOS's grub config with updates enabled. Rename its extension and paste it into the efi partition.

PixelOFC, 2020-05-30:
i put this using sudo edit-grub-config or on my linux grub config /boot/grub/?

khanhtranngoccva, 2020-05-30:
The efi partition of the Chrome OS disk, not your linux one!

PixelOFC, 2020-05-30:
ok ill try

khanhtranngoccva, 2020-05-30:
Ahhh, you use edit-grub-config on Chrome OS or edit the grub.cfg file on the ChromeOS' efi partition, DON'T use edit-grub-config on the linux used to install ChromeOS!

Skawtnyc, 2020-05-30:
Something to keep in mind: Even on chromebooks, triggering the update in Settings doesn't always work. The solution for me has been to reboot, log in, and before doing ANYTHING else, trigger the update from settings. Then don't touch it and let it finish until it tells you to reboot.

PixelOFC, 2020-05-31:
tbh i didnt tried yet, too much school shit

PixelOFC, 2020-05-31:
didnt worked, i tried normally and using @Skawtnyc method

PixelOFC, 2020-05-31:
any suggestion @sebanc ?

sebanc, 2020-06-01:
Could you post your linux grub config ?

khanhtranngoccva, 2020-06-03:
Mine failed as well.
[grub.txt](https://github.com/sebanc/brunch/files/4721167/grub.txt)

khanhtranngoccva, 2020-06-03:
chromeos hatch r80. My config boots, but no updates.

sebanc, 2020-06-03:
hatch recovery image is not yet supported for updates

khanhtranngoccva, 2020-06-03:
I really tried to use first-class Chromebook images, but it's a shame it won't work.

PixelOFC, 2020-06-03:
@sebanc im using Rammus r80, i dont have any update (stable is r81 if ikr)

[grub.txt](https://github.com/sebanc/brunch/files/4725130/grub.txt)

PixelOFC, 2020-06-03:
worked i was puting the option in the wrong location

closed 2020-06-03 by PixelOFC