sebanc/brunch #232: Remove brunch, and get windows 10 back:

itsmeokaoka, 2020-05-30:
Hi @sebanc ,
I love your work, but I want to reverse the progress and get windows 10 back.
I'm using a Medion Akoya laptop.
 Already tried a bootable win10 usb + tried get to boot options/bios, but I can only get it to show the grub boot, which just starts chrome os right away.
You know what to do?.

sebanc, 2020-05-30:
Hi,

When you say "I can only get it to show the grub boot", are you talking of the usb grub or the hdd grub ?

Skawtnyc, 2020-05-30:
The easiest way to do it would be to simply boot using the Windows 10 USB, use the command prompt in it, and wipe all of the partitions using diskpart. Then install Windows as normal to the empty hard drive.

itsmeokaoka, 2020-05-31:
Sorry, I ment that I only have: hdd grub menu.
Just looking for a way to boot from the usb

sebanc, 2020-05-31:
Your bios should be able to boot the windows 10 usb (brunch/grub changes nothing related to this). Have you set usb first in boot order ? How have you created the windows 10 usb install media ?

itsmeokaoka, 2020-06-01:
Hi again,
Can't even get to BIOS, If I could just do that, I would be good from there.
Tried every key.
I know my problem is at my end, but im really lost in this.
Im used to do "Rufus" win 10 usb, so I know that it is bootable.

sebanc, 2020-06-01:
When you see grub, press c and type "ls", identify your windows disk (something like "(hd0,1)") and then type:
`chainloader (hd0,1)+1` (adapt with your windows disk reference)

itsmeokaoka, 2020-06-01:
Nothing happens when pressing c. Guess I have to take out the hdd somehow and format it with the help of another pc

sebanc, 2020-06-01:
The issue you are having is very weird, pressing "c" in grub (or equivalent key for "c" with an english keyboard) should bring a grub prompt.

Indeed, if it does not work I see no other solution than reinstalling windows on an other pc.

closed 2020-06-26 by sebanc