sebanc/brunch #233: Internal microphone on Asus UX301 not working

lucderheld, 2020-05-30:
Sound is working, but internal mic is not working.
`localhost / # amixer
Simple mixer control 'Master',0
  Capabilities: pvolume pvolume-joined pswitch pswitch-joined
  Playback channels: Mono
  Limits: Playback 0 - 100
  Mono: Playback 91 [91%] [on]
Simple mixer control 'Capture',0
  Capabilities: cvolume cvolume-joined cswitch cswitch-joined
  Capture channels: Mono
  Limits: Capture 0 - 100
  Mono: Capture 80 [80%] [on]`

I tried the 'toggle' suggestions but it didnt fix the problem.

sebanc, 2020-06-02:
Hi, indeed I have many microphone not working issues, most of those grouped in #56.

Your laptop model seems interesting to try something, by any chance, are you on the telegram croissant/brunch group ? I would like to try an experimental build but I don't really like the idea of posting it here.

sebanc, 2020-06-04:
Hi,

I eventually put the test build here if you want to give it a try:
https://github.com/sebanc/brunch-testing/releases

closed 2020-06-26 by sebanc