sebanc/brunch #234: storage managment

au79912, 2020-05-31:
I have a 500gb HD and when I open the storgage mangment it gives me 251gb of avialable space and the rest of thee 249 gb is taken by system
is this normal?

sebanc, 2020-05-31:
Hi,

Could you post the output of "df -h" in crosh shell ?

au79912, 2020-05-31:
[image]

sebanc, 2020-05-31:
Ok, so the issue is that the data partition is only taking 284gb out of your 500 gb hdd.

How did you install ? from chromeos or from a linux live flashdrive ?

au79912, 2020-05-31:
No, I made a chrome os ISO in windows and then flashed it on a usb by Rufus
[image]

sebanc, 2020-05-31:
but how did you install on hdd ? or is it a usb hdd ?

au79912, 2020-05-31:
> but how did you install on hdd ? or is it a usb hdd ?

i installed it by running
 sudo chromeos-install -dst /dev/sda
on the terminal while booted from the usb

sebanc, 2020-06-01:
It looks like there is an issue with my install script, chromeos should have taken all the space on the disk instead of just 284 gb. Do you remember which brunch version you used when you installed chromeos on the disk ?

au79912, 2020-06-01:
brunch 81v4 but updated it to 83 beta about a few days ago

sebanc, 2020-06-01:
Updates do not modify the partition table but brunch 81v4 is probably recent enough that the issue is still present.

Could you backup your important data, reboot, press CTRL+ALT+F2 before login, login as "root" and run "sudo resize-data" ?

After rebooting, could you post the output of "sudo cgpt show /dev/sda" ?

au79912, 2020-06-01:
[image]

sebanc, 2020-06-01:
What's the output of: cat "/sys/block/sda/size" ?

au79912, 2020-06-01:
625142448

sebanc, 2020-06-01:
What's weird is that your hdd reports a size of 625142448 blocks of 512 bytes size which is around 300gb. Based on that the installer created an 300gb size partition table.

I don't understand why your 500gb hdd would report itself as 300gb but that's the origin of the issue.

You might try to reset your hdd through gparted on a linux live distro, verify that it is recognized as 500gb and then reinstall chromeos.

closed 2020-06-02 by au79912