sebanc/brunch #235: Enable Developer Mode

JakunGamer, 2020-05-31:
So I wanted to activate Developer Mode for tinkering a bit with the system ( That is what I always do ). Anyway that I can do that? When first boot I already choose Enable Developer Mode and set the root password but it didn't work.

sebanc, 2020-05-31:
Hi,

With brunch, chromeos is in developer mode by default without selecting any additional debugging option and the rootfs is already writable.

What are you trying to achieve ?

JakunGamer, 2020-06-04:
> Hi,
> 
> With brunch, chromeos is in developer mode by default without selecting any additional debugging option and the rootfs is already writable.
> 
> What are you trying to achieve ?

oh ok

closed 2020-06-04 by JakunGamer

dheerau, 2021-11-02:
how do i off the developer mode with brunch chrome os

ankitdasuni, 2022-01-24:
> Hi,
> 
> With brunch, chromeos is in developer mode by default without selecting any additional debugging option and the rootfs is already writable.
> 
> What are you trying to achieve ?

how to install vs code please help