sebanc/brunch #236: Chrome Os not booting HDD on Linux.

dumanOktay, 2020-05-31:
I can boot Chrome Os from my USB but can't boot from my HDD. 

when I select ChromOs from grub stopping on a black screen with no errors.

sebanc, 2020-05-31:
Hi,

I guess you are dual booting with linux. Could you try replacing your linux distro grubx64.efi (usually located in /boot/efi/EFI/<your_linux_distro>) with the brunch one located in /efi/boot in partition 12 of your usb ?

dumanOktay, 2020-05-31:
won't it break my system completely? :)

sebanc, 2020-05-31:
It should not but back up your ditro's grubx64.efi first and if it does not work from a live linux and change it back.

dumanOktay, 2020-06-02:
Thank you @sebanc  it works :)

khanhtranngoccva, 2020-06-02:
Does your image support legacy boot?

dumanOktay, 2020-06-02:
its support but it is on UEFI mode

closed 2020-06-07 by dumanOktay

srulikuk, 2020-07-26:
@sebanc 
Please pin this issue or add it to setup instructions, been troubleshooting for hours until i found this.. almost opened a ticket 5 times...