sebanc/brunch #237: Data partition resize - chromeos.img disk image resize

santhoshr, 2020-05-31:
Hi 

I found a way to resize img file, without triggering repair or restore. Including it for others, @sebanc you can add it to readme, wiki or link to this issue, if you think it will be useful.

You need linux and sudo, and developer mode in Chrome OS.

Take backup of chromeos image file before attempting.

In Linux,
1. Install kpartx and cloud-utils for growpart 
2. Increase the image using count, each count is 1MB so 1024 count is 1GB

> dd if=/dev/zero of=/path/to/chromeos.img bs=1M count=1024 oflag=append conv=notrunc status=progress

3. Fix GPT headers by running fdisk, it will complain about backup GPT table, and **type 'w'** in the fdisk command prompt to fix the headers,
> fdisk /path/to/chromeos.img
4. run below command to create loop device, it will print out loop device with partitions, note loop device id, usually /dev/loop0
> kpartx /path/to/chromeos.img
5. grow the data partition using below command, based on loop device identified previously and partition 
> growpart /dev/loop0 1

In Chrome OS,
6. Reboot to Chrome OS, hopefully it will boot, go to crosh -> shell -> root prompt (sudo su), type below command to find loop device for data partition (mine is /dev/loop15p1)
> df -h | grep /mnt/stateful_partition$
7. Run the below command to resize, based on previously identified loop partition device,
> resize2fs /dev/loop15p1

Congrats you have resized your Chrome OS data partition!

closed 2020-06-26 by sebanc

farhansj, 2023-11-12:
In Ubuntu (Linux), the command for the step 4 is `kpartx -av /path/to/chromeos.img`.

You can also press CTRL+ALT+F2 and login as root to execute step 6 and 7.

Regards

Andrew-Sor, 2024-12-28:
> 7\. run below command to create loop device, it will print out loop device with partitions, note loop device id, usually /dev/loop0
> 
> > kpartx /path/to/chromeos.img

Hello, I also want to extend the Chrome OS image, but after entering this command I get `/dev/mapper/control: open failed: Permission denied
Failure to communicate with kernel device-mapper driver.
Incompatible libdevmapper 1.02.175 (2021-01-08) and kernel driver (unknown version).
Bad address
can't set up loop`. Is there a way to fix this?