sebanc/brunch #238: error: no suitable video mode found. Booting in blind mode

pwnrepeat, 2020-05-31:
Hi sebanc,
I get this error when I try to boot into Chrome OS on my laptop:
error: no suitable video mode found. 
Booting in blind mode
If you could help me out, that would be great :D 
Thanks,
pwnrepeat

pwnrepeat, 2020-05-31:
Update: I extracted the mbr.tar.gz and ran the setup again, I am now getting a log

pwnrepeat, 2020-05-31:
However It gets past a certain point then restarts...

sebanc, 2020-05-31:
Do you have a laptop or a desktop ?

Try to verify in your bios if boot mode is set to UEFI Hybrid (with CSM).

Else, you can try setting the video mode in kernel command line by adding something like "video=1920x1080-32"

pwnrepeat, 2020-05-31:
how would i set the video mode

pwnrepeat, 2020-05-31:
I have a laptop

pwnrepeat, 2020-05-31:
my boot mode is set to EFI or it can be set to legacy

pwnrepeat, 2020-05-31:
It now boots but it restarts at a certain point

pwnrepeat, 2020-05-31:
https://gofile.io/d/icg4bV

pwnrepeat, 2020-05-31:
There is the video of it booting.

pwnrepeat, 2020-05-31:
> Do you have a laptop or a desktop ?
> 
> Try to verify in your bios if boot mode is set to UEFI Hybrid (with CSM).
> 
> Else, you can try setting the video mode in kernel command line by adding something like "video=1920x1080-32"

I have a laptop

pwnrepeat, 2020-05-31:
I updated to the latest testing release, chromeos is not found...

pwnrepeat, 2020-05-31:
I am now going to try the latest stable release, as I was using R80 to start out with.

sebanc, 2020-05-31:
I cannot open the link you provided for the video, could you try to attach it here as zip ?

Also, what's your laptop model ? Did you try to update to r83 ?

pwnrepeat, 2020-05-31:
yeah
ok

pwnrepeat, 2020-05-31:
my laptop is a samsung r530

pwnrepeat, 2020-05-31:
i updated to r83
and i got chromeos is not found.

pwnrepeat, 2020-05-31:
im using samus

pwnrepeat, 2020-05-31:
If you have discord, can you dm me at pwnrepeat#0001

sebanc, 2020-05-31:
Sorry I mixed 2 issues, your laptop never booted brunch right ?

pwnrepeat, 2020-05-31:
it has now

pwnrepeat, 2020-05-31:
But it restarts after 15 - 20 seconds

pwnrepeat, 2020-05-31:
https://youtu.be/xL6f9DelBfE

sebanc, 2020-05-31:
The chromeos images prebuilt by google contains cpu instructions which are not present in core 2 duo cpu. It cannot boot further than this.

You can try:
- Fydeos is built from chromiumos source so it might work.
- orl an android-x86 fork, someone on the Telegram group recently mentioned "PrimeOS Classic".

pwnrepeat, 2020-05-31:
Ah ok

pwnrepeat, 2020-05-31:
Thanks alot.

closed 2020-05-31 by pwnrepeat

hacker1024, 2020-08-26:
I'm getting this on boot, and no kernel logs show or anything - but soon after, the UI opens, and everything seems fine. Does anyone know what could be going on here?

I am using r84, so that may be it