sebanc/brunch #239: HP pavilion x360 i5-10210u rtl8822ce bluetooth wifi combo coexistence

senssopy, 2020-05-31:
HP pavilion x360 i5-10210u rtl8822ce bluetooth wifi combo coexistence

Hi,

I can use wifi because of your driver update.
But when I turned on the bluetooth, it didn't work.
Next, I turned off the wifi, turned on the bluetooth. The bluetooth is work.

Just not simultaneously.

I googled...

My wifi and bluetooth model is rtl8822ce, it is just one, a combo model.

I think the below data is not perfect to fix my issue.
Because it is not provided by the manufacture.
Just it was made by people, for free computer valenteers.

Anyrate, if you could insert next data, I will happily try it if it work or not.

Data 1.
https://patchwork.ozlabs.org/project/openwrt/patch/20191021191308.16311-2-mail@david-bauer.net/

Data 2.
https://patchwork.kernel.org/patch/11365833/

Data 3.
https://aur.archlinux.org/packages/rtl88x2ce-dkms/

(Data 3.
https://drive.google.com/file/d/1mlYnkQ2eiQ5Y3LI1REa20XI-XJBVhde5/view?usp=sharing

)

Or you can order me to do something, using terminal, modify grub... etc.
I can install and use crouton.

sebanc, 2020-06-01:
Hi, could you try the new release in "brunch-testing" repository ?

senssopy, 2020-06-05:
Thx for your fast respond.
I tried it.
Bluetooth is normal.
Wifi is not work.
Tethering is ok.
Here's a dmesg.

https://drive.google.com/file/d/1UzyA_x7aj5alvpSzAUDKJYlZJhMk1E2C/view?usp=sharing

sebanc, 2020-06-05:
I cannot access the dmesg, I think the file is still set on private mode.

Also, could you post the output of "sudo rfkill list all" ?

senssopy, 2020-06-05:
Ah!

1. dmesg

https://drive.google.com/file/d/1UzyA_x7aj5alvpSzAUDKJYlZJhMk1E2C/view?usp=sharing

2. rfkill

senssopy, 2020-06-06:
Sorry.
Before, I went websurfing, I knew just my wifi driver is rtl8822ce.

But WNFT-237ACN(bt) is a real my wifi&bluetooth module.

https://www.sparklan.com/product/wnft-237acnbt-rtl8822ce-mu-mimo-m-2-module/

rtl8822ce is just a chip.
(Indeed, the driver of rtl8822ce does not exist in realtek homepage.)

senssopy, 2020-06-06:
How about this one?
"coexistence setting"
from here

https://github.com/juanro49/rtl88x2ce-dkms

https://drive.google.com/file/d/1uXBxn2rJwO0I3oZ7YIY7nmUh4CUXFxwc/view?usp=sharing

sebanc, 2020-06-06:
Does your wifi still works without bt enabled ?

Could you try the same with a recent live usb linux distro an check if you have the same issue ? In the last testing release, I had backported the kernel driver from linux 5.6.16.

senssopy, 2020-06-06:
Before I check if wifi still works, I installed ubuntu usb.. ^ ^
Soon, I will notice you what is the truth.

I tried ubuntu-18.04.4-desktop-amd64, live usb, not install.
Wifi and bluetooth work simultaneously.
Specailly, a rtl8822ce works like rtl8822cu. (I don't know what is rtl8822cu.)

Here's a dmesg by ubuntu.

https://drive.google.com/file/d/1oNW5FksR7aTVZsjhU1v4F0hMtU1ftHf1/view?usp=sharing

senssopy, 2020-06-06:
r81(r80 also), rammus, newtestingbrunch0604.

When a bluetooth turns off, a wifi doesn't work.

I typed a rtl8822cu in github.
There are some project to improve(maybe simultaneously work) bt part of the rtl8822cu.

https://github.com/PLVision/linux-firmware/commit/180e2b46aa25595ee5462e3690f542085fe5f81f

senssopy, 2020-06-08:
Awesome ^ ^ !
A newtesting ver. enables wifi and bluetooth to work simultaneously.
You're competent. Awesome. Thx!!!

God bless you ^ ^

I use it useful in corona time. IT is useful It. Thx ^ ^ .

closed 2020-06-08 by senssopy

sebanc, 2020-06-08:
Nice to see that you tested it (even though I forgot to update the issue) and that it works