sebanc/brunch #241: CPU frequency stuck at 1800Mhz

Zatfer17, 2020-05-31:
Hi, I installed today Rammus 80 img on my t480s. I've been enjoying it but the battery life is not the best. In particular I noticed that the cpu consistantly sits on 1800mhz. Any thoughts on how to solve this?

sebanc, 2020-06-01:
Are you using a recent brunch version ? I remember having a similar issue on old builds

Zatfer17, 2020-06-01:
Yup, I'm on brunch r83 k4.19 testing 20200528

sebanc, 2020-06-01:
No, it will be the same, the issue was in performance mode instead of ondemand in old releases but I fixed it in recent ones. Just in case, could you post the output of:
cat /sys/devices/system/cpu/cpu*/cpufreq/scaling_governor

Zatfer17, 2020-06-01:
I eventually figured out that I was gathering this info in the Linux (beta) terminal, so it was completely different from the output of the shell. I can confirm you that the output you are requesting is powersave and that the cpu idles correctly at 800mhz. Last time I used ChromeOs there was no Crostini and just Crouton so I got kinda confused. Thank you so much!

closed 2020-06-01 by Zatfer17