sebanc/brunch #242: issues after reboot

tungmeister, 2020-05-31:
@sebanc  I've noticed every time I switch back to windows after a reboot from Chrome OS the time is behind by one hour. Also if I try and switch to ubuntu (tripple boot) it fails to boot the first time (sits at the loading animation indefinitely) through a reboot cures this.  The ubuntu issue only happens whne I've been in Chrome OS. Any ideas?

khanhtranngoccva, 2020-06-01:
Clock? Chrome OS corrupts the BIOS and CMOS data, since it's trying to communicate with their own proprietary BIOS. This resets the time zone, and sometimes the UTC time itself.  I've seen some BIOS malfunction in my computer as well.

sebanc, 2020-06-01:
Have you tried the latest r83 brunch ? This should not happen anymore.

petertriho, 2020-06-01:
Use option 2 https://www.howtogeek.com/323390/how-to-fix-windows-and-linux-showing-different-times-when-dual-booting/

midi1996, 2020-06-04:
it's probably windows UTC issue, pretty known when dualbooting.

closed 2020-06-26 by sebanc