sebanc/brunch #243: Boot Loop

memocl, 2020-06-01:
updated brunch
options=updates_enabled
installed the latest chrome os release 
Boot
Unable to get to the Chrome Logo.
Loops back to boot

sebanc, 2020-06-01:
Hi, what's your laptop cpu and which recovery are you using ?

memocl, 2020-06-01:
Acer Aspire V with  Intel Core i7-3537U 
rammus recovery

sebanc, 2020-06-01:
From r83, google builds rammus with cpu instructions which are unsupported by intel processors <= gen 3.
To be able to update, you have to switch to samus recovery image.

memocl, 2020-06-01:
Will do and comment back.
Thanks in advance.
And thanks a bunch for Brunch

DennisLfromGA, 2020-06-01:
I can confirm this also.

I started with an M80 rammus recovery image on an old Acer C7 and all worked fine until the upgrade to M83 - boot loop.

I then used an M80 samus recovery image and once I enabled upgrades it upgraded just fine - NO boot loop.

I should've checked my CPU gen. *before* I dove in - lesson learned.

'brunch` is *awesome*, thanx so much.

memocl, 2020-06-01:
Resolved.
Worked as @DennisLfromGA  said.
Thankyou @sebanc

closed 2020-06-02 by sebanc