sebanc/brunch #244: FR: Use partition 8,9, or 10 for brunch framework

DennisLfromGA, 2020-06-01:
This is just a feature request.

Some like to use partition 7 for dual-booting via chrx or crouton chroot storage, etc. which is safe from powerwashes.

The `chrx` script doesn't allow for using a different partition so doing so would require a manual effort and knowledge of partitions, sizing, etc. which a lot of users don't have.

I don't know how big an effort it would be to move the brunch framework tho so if this isn't feasible I understand.

Thanx,
-**DennisLfromGA**

**P.S.** Thanx a bunch for brunch.

sebanc, 2020-06-04:
Hi,

I have two issues here:
- Changing brunch partition mechanism is quite complicated,
- Even if I would change brunch partition mechanism, chrx will not work (anyway in my opinion the brunch dual boot procedure is more appropriate).

I did not know some used partition 7 to store crouton chroots which indeed makes sense to me. However, I think it would be easier for me to create a script to allow the use of another partition for crouton backups.

Could you point me to the script you currently use in chromeos to allocate space on partition 7 ?

DennisLfromGA, 2020-06-04:
No worries, there are probably only a few that would want to dual-boot or, like me, store crouton chroots on custom partitions so it's definitely not worth the effort I'm sure. I can imagine it would be a  pretty complicated task and then you'd run into pre & post versions which would further complicate things.

As far as scripts to allocate space on partition 7 I use `chrx` now but years ago I was using a script called `mkpart.sh` written by a Googler named drinkcat@ expressly for crouton that allocated space on partition 13 for chroots -
- crouton in a separate partition by drinkcat · Pull Request #706 · dnschneid/crouton
https://github.com/dnschneid/crouton/pull/706  

That was before frecon came about and precluded being able to switch virtual terminals which broke it.

Anyway, thanx for even considering it, the FR was kind of a whim that I had and definitely not something I was counting on.

*Thanx again,*
-**DennisLfromGA**