sebanc/brunch #245: FR: Check $EDITOR variable for edit-grub-config

DennisLfromGA, 2020-06-01:
This is just a feature request.

Some like to use an editor of their choice for editing scripts & configs like the grub.cfg file, etc.
I would be handy for the `/usr/sbin/edit-grub-config` to check this variable and use it instead of always using `nano`. 

I was a bit surprised when the `grub.cfg` file opened in `nano` instead of `vi` until I finally figured out why. ;-)

Thanx,
-**DennisLfromGA**

**P.S.** Thanx a bunch for brunch.

sebanc, 2020-06-04:
Hi,

I have added this option in the brunch unstable release.
https://github.com/sebanc/brunch-testing/releases

Use: sudo EDITOR=vi edit-grub-config

DennisLfromGA, 2020-06-04:
*Awesome!* Thanks for not only considering this but implementing it. 👊

DennisLfromGA, 2020-06-04:
I just now updated to framework [brunch_r83_k4.19_unstable_20200604](https://github.com/sebanc/brunch-testing/releases/tag/r83-k4.19-unstable-20200604).

The $EDITOR variable works as described and everything else does too. 

Thanx so much,
-**DennisLfromGA**

closed 2020-06-26 by sebanc