sebanc/brunch #247: Huawei E3372h-153 USB Modem not working

OkameProductions, 2020-06-02:
Because of my problems with WiFi + BT (will report later), I tried using a USB modem to connect to the internet before falling back to a phone with USB tethering (which worked smoothly).

## Expected behavior
Upon plugging in the modem, ChromeOS should let me connect to a mobile connection through the WiFi tab.

## Actual behavior
Upon plugging in the modem, ChromeOS detects a USB Ethernet connected, attempts connecting to it, but nothing is happening on the modem (still blinking for connection.)

OkameProductions, 2020-06-02:
Here is output from `dmesg` after plugging in the Huawei modem:
```
[ 5421.722134] usb 2-1.2: new high-speed USB device number 9 using ehci-pci
[ 5421.805919] usb 2-1.2: New USB device found, idVendor=12d1, idProduct=14fe, bcdDevice= 1.02
[ 5421.809293] usb 2-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5421.812550] usb 2-1.2: Product: HUAWEI_MOBILE
[ 5421.815899] usb 2-1.2: Manufacturer: HUAWEI_MOBILE
[ 5421.819173] usb 2-1.2: SerialNumber: 0123456789ABCDEF
[ 5421.853635] usb-storage 2-1.2:1.0: USB Mass Storage device detected
[ 5421.858166] scsi host6: usb-storage 2-1.2:1.0
[ 5422.318142] usb 2-1.2: USB disconnect, device number 9
[ 5422.747046] usb 2-1.2: new high-speed USB device number 10 using ehci-pci
[ 5422.829147] usb 2-1.2: New USB device found, idVendor=12d1, idProduct=1506, bcdDevice= 1.02
[ 5422.831870] usb 2-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 5422.833608] usb 2-1.2: Product: HUAWEI_MOBILE
[ 5422.835413] usb 2-1.2: Manufacturer: HUAWEI_MOBILE
[ 5422.906744] usb-storage 2-1.2:1.3: USB Mass Storage device detected
[ 5422.910765] scsi host6: usb-storage 2-1.2:1.3
[ 5422.915172] usb-storage 2-1.2:1.4: USB Mass Storage device detected
[ 5422.916835] scsi host7: usb-storage 2-1.2:1.4
[ 5423.067130] usbcore: registered new interface driver option
[ 5423.068648] usbserial: USB Serial support registered for GSM modem (1-port)
[ 5423.070731] option 2-1.2:1.0: GSM modem (1-port) converter detected
[ 5423.072912] usb 2-1.2: GSM modem (1-port) converter now attached to ttyUSB0
[ 5423.074602] option 2-1.2:1.1: GSM modem (1-port) converter detected
[ 5423.076235] usb 2-1.2: GSM modem (1-port) converter now attached to ttyUSB1
[ 5423.092737] usbcore: registered new interface driver cdc_ncm
[ 5423.100641] usbcore: registered new interface driver cdc_wdm
[ 5423.201694] huawei_cdc_ncm 2-1.2:1.2: resetting NTB format to 16-bit
[ 5423.204676] huawei_cdc_ncm 2-1.2:1.2: MAC-Address: 00:1e:10:1f:00:00
[ 5423.207873] huawei_cdc_ncm 2-1.2:1.2: setting rx_max = 16384
[ 5423.218584] huawei_cdc_ncm 2-1.2:1.2: NDP will be placed at end of frame for this device.
[ 5423.220937] huawei_cdc_ncm 2-1.2:1.2: cdc-wdm1: USB WDM device
[ 5423.223965] huawei_cdc_ncm 2-1.2:1.2 wwan0: register 'huawei_cdc_ncm' at usb-0000:00:1d.0-1.2, Huawei CDC NCM device, 00:1e:10:1f:00:00
[ 5423.226656] usbcore: registered new interface driver huawei_cdc_ncm
[ 5423.617806] IPv6: ADDRCONF(NETDEV_UP): veth_wwan0: link is not ready
[ 5423.628499] arc_wwan0: port 1(veth_wwan0) entered blocking state
[ 5423.629935] arc_wwan0: port 1(veth_wwan0) entered disabled state
[ 5423.631661] device veth_wwan0 entered promiscuous mode
[ 5423.651890] wwan0: renamed from peer_wwan0
[ 5423.661586] IPv6: ADDRCONF(NETDEV_UP): wwan0: link is not ready
[ 5423.663475] IPv6: ADDRCONF(NETDEV_CHANGE): wwan0: link becomes ready
[ 5423.665693] IPv6: ADDRCONF(NETDEV_CHANGE): veth_wwan0: link becomes ready
[ 5423.667838] arc_wwan0: port 1(veth_wwan0) entered blocking state
[ 5423.669677] arc_wwan0: port 1(veth_wwan0) entered forwarding state
[ 5423.921107] scsi 6:0:0:0: CD-ROM            HUAWEI   Mass Storage     2.31 PQ: 0 ANSI: 2
[ 5423.924626] sr 6:0:0:0: Power-on or device reset occurred
[ 5423.929516] sr 6:0:0:0: [sr1] scsi-1 drive
[ 5423.931573] sr 6:0:0:0: Attached scsi CD-ROM sr1
[ 5423.985534] scsi 7:0:0:0: Direct-Access     HUAWEI   TF CARD Storage  2.31 PQ: 0 ANSI: 2
[ 5423.988254] sd 7:0:0:0: Power-on or device reset occurred
[ 5423.993616] sd 7:0:0:0: [sdb] Attached SCSI removable disk
```

And here's the screenshot of the network tab:

[image]

sebanc, 2020-06-03:
Looks like this usb modem has low compatibility with linux:
https://askubuntu.com/questions/757638/can-not-connect-huawei-e3372-modem-on-ubuntu-15-10-please-help

As far as I can see all kernel requirements are already implemented, you can try following the directions that are on the above website but it seems quite complicated.

OkameProductions, 2020-06-03:
The modem works fine under my arch install (thanks to modem-manager), and
native ChromeOS should have support for it too, which is why it puzzles me.
I'll try following the instructions later. Thanks for the help nonetheless.

sebanc, 2020-06-03:
http://www.chromium.org/chromium-os/how-tos-and-troubleshooting/get-a-3g-modem-running

```
What if the modem is not seen by Modem Manager?
- make sure kernel has proper drivers
- do you need to run modem-modeswitch?
- do you need to eject the modem?
- Can you write some udev rules to automatically eject/modeswtich the modem?
```

You are in case 2: do you need to run modem-modeswitch?

OkameProductions, 2020-06-03:
No, the modem switches automatically..
The same under ChromeOS: it automatically switches modes, but instead of
being able to connect to the mobile network, I'm instead presented with an
Ethernet network, which doesn't seem to connect.

sebanc, 2020-06-03:
Ah ok I did not think it would switch to modem mode automatically. Once it is displaying ethernet, have you tried connecting to this address under chrome to input your PIN : http://192.168.3.1 ?

OkameProductions, 2020-06-03:
The modem doesn't have a PIN set up, and ChromeOS couldn't connect to the Ethernet.

OkameProductions, 2020-06-03:
It might be because the Ethernet port in this laptop is broken, but I don't know.

sebanc, 2020-06-03:
Sorry but I don't know much about those adapters, from what I read all necessary kernel modules are included. I have no idea what could be missing.

OkameProductions, 2020-06-03:
This might be interesting.
```
chronos@localhost / $ sudo modem status

Modem /org/freedesktop/ModemManager1/Modem/0:
  GetStatus:
    state: 3
  Properties:
    Sim: /org/freedesktop/ModemManager1/SIM/0
    SupportedCapabilities: 4
    CurrentCapabilities: 4
    MaxBearers: 2
    MaxActiveBearers: 2
    Manufacturer: huawei
    Model: E3372
    Revision: 21.318.01.00.55
    CarrierConfiguration: 
    CarrierConfigurationRevision: 
    HardwareRevision: 
    DeviceIdentifier: 53a28ed378f84c7373dce2fdaf92d7530bb47777
    Device: /sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2
    Drivers: option1, huawei_cdc_ncm
    Plugin: Huawei
    PrimaryPort: ttyUSB0
    Ports: ttyUSB0, 3, ttyUSB1, 3, wwan0, 2
    EquipmentIdentifier: <no>
    UnlockRequired: 1
    UnlockRetries: 3, 3, 10, 10
    State: 3
    StateFailedReason: 0
    AccessTechnologies: 0
    SignalQuality: 0, false
    PowerState: 3
    SupportedModes: 2, 0, 4, 0, 8, 0, 14, 0
    CurrentModes: 14, 0
    SupportedBands: 0
    CurrentBands: 0
    SupportedIpFamilies: 1
  3GPP:
    Imei: <no>
    RegistrationState: 4
    OperatorCode: 
    OperatorName: 
    EnabledFacilityLocks: 0
    SubscriptionState: 0
    EpsUeModeOperation: 4
    InitialEpsBearer: /
  CDMA:
  SIM /org/freedesktop/ModemManager1/SIM/0:
    SimIdentifier: <no>
    Imsi: <no>
    OperatorIdentifier: 26006
    OperatorName: nc+
```
Censored sensitive information with `<no>`s.

OkameProductions, 2020-06-03:
I couldn't run any `mm-*` commands though, such as `mm-connect` or `mm-enable`, neither as user or root with sudo.

sebanc, 2020-06-03:
The only 2 things I see:
- Drivers: option1, huawei_cdc_ncm => some articles I read mentioned another module: cdc_ether. You can try blacklisting this module.
- UnlockRequired: 1 => seems suspicious but no idea what to do about this.

OkameProductions, 2020-06-03:
I really hope this isn't another dualboot-related issue, like WiFi and Bluetooth before on this machine, although it shouldn't be since I was also rebooting with this stick on and it wouldn't budge.
This stick isn't a HiLink stick. If I run `ls /dev/ttyUSB*`, I can see the modem's serial interfaces being there. ModemManager or wvdial would take those and establish a stable connection. This doesn't happen under ChromeOS.
We could try adding `wvdial` into ChromeOS for manual connections, but I highly doubt its gonna help with the native GUI and ModemManager issues.
Other bet is trying to add ChromiumOS's `flimflam` or its `ModemManager` packages.
I'll try blacklisting the module.
Btw, I think it's `module_blacklist=huawei_cnc_ncm`, right?

OkameProductions, 2020-06-03:
About the `UnlockRequired`, it also seems suspicious to me, but the SIM doesn't have any PIN set up??

OkameProductions, 2020-06-03:
> I couldn't run any `mm-*` commands though, such as `mm-connect` or `mm-enable`, neither as user or root with sudo.

I also can't run `usb_modeswitch`.
[image]
[image]

OkameProductions, 2020-06-03:
GOOD NEWS!
Blacklisting the driver did help, I'm connected successfully without problems.

[image]
[image]

closed 2020-06-03 by OkameProductions