sebanc/brunch #248: Lenovo G570 | Problems with WiFi + BT (AR5B195) under ChromeOS

OkameProductions, 2020-06-02:
# WiFi
## Expected behavior
WiFi is available, all nearby networks should be available, and I should be able to connect to a WiFi network.

## Actual behavior
WiFi is reported as on and available, however no networks are shown, and attempting to add and connect to a network through "Add hidden network" fails to connect.

## Possible fixes?
Adding a proper Atheros AR9285/AR5B95 WiFi driver.

# Bluetooth
## Expected behavior
Bluetooth is enabled, and I can connect to a Bluetooth device such as a phone.

## Actual behavior
Bluetooth is fully disabled, without the ability to turn it on (switch is also disabled). The notifications panel is missing the Bluetooth pill.

## Possible fixes?
Adding a proper Atheros BT driver.

sebanc, 2020-06-02:
Hi,

Could you attach a dmesg to this issue ? (run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and attach dmesg.txt from your downloads folder to this issue).

Also, could you post the output of "sudo rfkill list all" ?

OkameProductions, 2020-06-02:
Yeah, I'll post it soon.
Right now I'm sitting on my Arch install.

OkameProductions, 2020-06-02:
Ok now that I'm actually in ChromeOS WiFi now works?? The networks are
available and I can connect to WiFi. Still no BT tho, will send reports
soon.

OkameProductions, 2020-06-02:
Done getting the outputs.

```
chronos@localhost / $ sudo rfkill list all
0: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
```

[Here is dmesg's output.](https://github.com/sebanc/brunch/files/4716481/dmesg.txt)

OkameProductions, 2020-06-02:
Ok, solved the WiFi mystery!
I had WiFi turned off in Arch Linux, and when turning it back on and rebooting to ChromeOS, it works! I'll try the same with Bluetooth.

OkameProductions, 2020-06-02:
Yup! That solved the issue for both cases.
[image]
The solution is to enable both in a different Linux distro (especially important in dualbooting!)
Thanks for the help, closing.

closed 2020-06-02 by OkameProductions