sebanc/brunch #249: How to dual boot ChromeOS with a MBR+BIOS Windows+Linux install?

khanhtranngoccva, 2020-06-02:


OkameProductions, 2020-06-02:
If you have GRUB2 installed into your MBR, you can make a ChromeOS image with Brunch on the Linux partition and boot from it.

I also recommend making the image with dd and mounting it with losetup (`sudo losetup -Pf /chromeos/chromeos-disk.img`), then using the loop device for installing ChromeOS onto the image.

GRUB2 menuentry:
```
menuentry "ChromeOS (Disk image)" {
	img_part=/dev/sda2
	img_path=/chromeos/chromeos-disk.img
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
	initrd (loop,gpt7)/initramfs.img
}
```

khanhtranngoccva, 2020-06-02:
I have linux's mint default grub installed.

khanhtranngoccva, 2020-06-02:
I installed the BIOS variant of Windows before installing Linux Mint, therefore my system now runs on BIOS Grub2. Is it possible to run?

OkameProductions, 2020-06-02:
Yeah, give it a shot.
I'm dualbooting Arch + Chrome OS under MBR so it should work for you too.

khanhtranngoccva, 2020-06-02:
The system returned "no such partition, you need to load kernel first" and failed to detect loop partition.

OkameProductions, 2020-06-02:
My menuentry is pointing at my Arch partition, which happens to point to `/dev/sda2`.

Change the `img_part` to the partition which has the IMG file, and change the `img_path` to point to your IMG's location on the partition.

khanhtranngoccva, 2020-06-02:
I did that, but upon typing the commands I observed that there is one (loop) entry but no (loop,gptX) at all. If I use EFI Grub2 from USB the loopback commands work normally.

OkameProductions, 2020-06-02:
You might need to `insmod part_gpt` in GRUB2.

khanhtranngoccva, 2020-06-02:
I have just found that out as well.

closed 2020-06-02 by khanhtranngoccva