sebanc/brunch #250: Internal mic is silent / can`t change flags

marrecar, 2020-06-02:
I managed to install Chrome OS yesterday, with a little trouble and everything works seemingly fine. Till now, when I have to use online conference call (University) and it seems that my (internal) microphone is very silent/quiet. I googled a bit and can't seem to find a clear solution, other than to enable a flag that allows to change the volume of microphone (which seems to be in Chrome OS 81). And the chrome flags don't work apparently, they are blank.
What can I do? How can I solve this issue?

petertriho, 2020-06-02:
I don't remember seeing that flag in v81, I'm on v83 now and the flag is definitely there. So try updating to v83?

marrecar, 2020-06-02:
> I don't remember seeing that flag in v81, I'm on v83 now and the flag is definitely there. So try updating to v83?

Can you tell me how do you see the flags? My list doesn't show up at all, it's empty and I can't turn on any flags.

petertriho, 2020-06-02:
chrome://flags/system-tray-mic-gain in your browser

marrecar, 2020-06-02:
Yeah, I used that one, but nothing shows up, a blank page.

petertriho, 2020-06-02:
Then it's not avaliable on v81, I'm on v83, so you probably need to be on that version to have it

marrecar, 2020-06-02:
Yeah, I think so too. And then a stupid question, how do I update to that version? I don`t want to enable auto update so I dont break it, but I dont really have idea where do I get the latest version.

petertriho, 2020-06-02:
Just enable auto update, update, then disable it

marrecar, 2020-06-02:
Alright, thanks, will do that, and get back with the results!

marrecar, 2020-06-02:
Nope, still the same. Updated the OS, updated the brunch framework, and still can't adjust my microphone's volume. There is no option to it.
And my flags still don't appear. The list doesn't show up, nor does the flag "system-tray-mic-gain" appear when I go directly to it.
the thing I just noticed now is that, when I plug out my headset, the option where I can see the microphone, as internal, is gone. My laptop has its own microphone and I don't see any option to adjust it. It's really bugging me and it's really important to me. Especially since I'm gonna have an oral exam in about 3 weeks and I would have to change the OS back to other one, if I can't fix it here with Chrome OS.

sebanc, 2020-06-02:
Hi, I guess your are using samus recovery, if that's the case flags can only be changed from guest mode or in dev channel.

Also this issue can probably be solved by changing the mic volume from "alsamixer" command in crosh shell.

marrecar, 2020-06-02:
Yes, I'm using samus recovery.
Tried with alsamixer, but for some reason I can't increase the volume. After switching the sound card, and switching to "all", I can't increase any of the options. The "default" sound card has only one option, which works, but only changes sound volume.
How would I change to dev channel, if it's possible at all? I went to settings and found the option to change, but it's greyed out, inaccessible.
Is it possible do add the flag manually?

sebanc, 2020-06-02:
You can try setting the flag from guest mode.

Else, you can update to dev channel by running "sudo update_engine_client --channel=dev-channel -update" in crosh shell.

marrecar, 2020-06-02:
Flags appear on guest mode, but of course, doesn't save them.
Tried updating to dev channel by runing that command in shell, but didn't work.
Here's what happened:

> [0602/172105.945458:INFO:update_engine_client.cc(428)] Channel permanently set to: dev-channel
[0602/172105.945766:INFO:update_engine_client.cc(478)] Forcing an update by setting app_version to ForcedUpdate.
[0602/172105.945794:INFO:update_engine_client.cc(480)] Initiating update check.
[0602/172105.946712:INFO:update_engine_client.cc(509)] Waiting for update to complete.
[0602/172136.321716:ERROR:update_engine_client.cc(210)] Update failed, current operation is UPDATE_STATUS_IDLE, last error code is ErrorCode::kOmahaErrorInHTTPResponse(37)

sebanc, 2020-06-02:
It seems it updated the channel but not chromeos, which version are you on ?

marrecar, 2020-06-02:
Version 83.0.4103.77 (64-Bit)
Rebooted my laptop, still can't access flags. The channel says it's dev at the moment, but like you said, the os didn't change to dev.

sebanc, 2020-06-02:
Usualy this means that "options=enable_updates" is not set on the kernel command line but it's weird here as you have already updated to r83...

marrecar, 2020-06-02:
Okay, then that's the issue, since I removed that option, after I updated the system. Will try again with that option written.

marrecar, 2020-06-02:
Alright, that worked completely.
To write all again, in case someone needs it. I enabled auto update in kernel command line, rebooted, then used the update to dev channel command in shell and let it update. Then rebooted again and now I can access the flags. I changed the system-tray-mic-gain flag to enabled, restarted as offered, and now I can change the volume of my microphone. Tested it and it's now louder.
My current version is 84.0.4147.21 (Offizieller Build) dev (64-Bit).
Now I have a couple more questions, regarding this. Is it safe to stay on this build? Should I remove the auto update option again?
Also, regarding the boot up. Do I always have to go through the Grub screen and after that it goes through some " installation" or whatever it is? Sometimes it takes a minute or two until I boot to Chrome logo. If no, how do I disable that and speed up the boot?

Thank you guys again, really appreciate it!

sebanc, 2020-06-03:
I think that even with "options=enable_updates", the system will only update if you ask for it in settings so it is probably safe to leave it enabled.

The system rebuilds rootfs when there is an update or change to "options=...", it happened a lot as you changed options frequently but it should not happen now that your install is stable.

If you want, you can avoid grub by replacing "set timeout=2" by "set timeout=0" in grub config.

closed 2020-06-26 by sebanc