sebanc/brunch #251: Lenovo G570 | Compatibility results

OkameProductions, 2020-06-02:
# Table entry

Device | Type | Brunch version | Recovery | Status | Known issues | Notes
-- | -- | -- | -- | -- | -- | --
Lenovo G570 | Notebook | Brunch r83 k4.19 | nocturne | Working | Bluetooth doesn't work, Crostini not available because of lack of virtualization, camera doesn't work. | Microphone not tested, this laptop doesn't have (U)EFI, tested in a dualboot with Arch Linux.

# Known issues tests
* [x] Does the Android Subsystem (Android apps) run?
* [ ] Is Linux container install option available?
* [ ] Does Linux container install?
* [ ] Does Linux terminal run?
* [ ] Is Bluetooth working?
* [x] Is WiFi working?
* [ ] Is the camera working?
* [ ] Is microphone working? (Not tested)
* [x] Does trackpad and gestures work?

sebanc, 2020-06-02:
For crostini, is it not possible to enable VT-X in your bios ?

OkameProductions, 2020-06-02:
The Intel Pentium B960 itself doesn't support VT-x. Virtualization is enabled in BIOS.

OkameProductions, 2020-06-03:
The camera now decided to load, and it works.

closed 2020-06-29 by sebanc