sebanc/brunch #252: RT3290 Drivers for Bluetooth

Sahib911, 2020-06-02:
Can you please add drivers for RT3290. 
WiFi works fine but Bluetooth doesn't turn on .

OkameProductions, 2020-06-02:
My main laptop actually has the same card.
Good to know that WiFi works, will solve me some headaches.

Realtek Bluetooth drivers aren't in the Linux kernel, sadly.
There exists an open-source driver - `rtbth` - which is a Bluetooth driver for Realtek, and it may or may not work with ChromeOS.

@sebanc, could you try and incorporate `rtbth` into Brunch's kernel?
I'd be willing to test, but my main laptop's fan is fully dead, and it's pretty much unusable (overheating issues).
@Sahib911, if @sebanc adds `rtbth` into the kernel, could you test it?

Sahib911, 2020-06-03:
If added , I would happily test it .

sebanc, 2020-06-04:
Hi,

I added the rtbth driver in an unstable build available below (use "options=rtbth" to activate it)
https://github.com/sebanc/brunch-testing/releases

Could you try it ?

Sahib911, 2020-06-04:
Hello , I tried the unstable build but Bluetooth still won't turn on .

sebanc, 2020-06-04:
Could you attach a dmesg ? (run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and attach dmesg.txt from your downloads folder.

Sahib911, 2020-06-05:
[dmesg.txt](https://github.com/sebanc/brunch/files/4733881/dmesg.txt)
Here

sebanc, 2020-06-05:
Could you try to add "modprobe bluetooth" before "insmod /lib/modules/rtbth.ko" in /etc/init/rtbth.conf ?

From:
```
start on stopped udev-trigger

script
	insmod /lib/modules/rtbth.ko
end script
```

To:
```
start on stopped udev-trigger

script
	modprobe bluetooth
	insmod /lib/modules/rtbth.ko
end script
```

Sahib911, 2020-06-05:
Whenever I try to access the file it says permission denied. Can you explain , how to open the file .

sebanc, 2020-06-05:
From crosh shell, run "sudo nano /etc/init/rtbth.conf" to modify the content of the file.

Sahib911, 2020-06-05:
[dmesg.txt](https://github.com/sebanc/brunch/files/4734598/dmesg.txt)
Still no luck

sebanc, 2020-06-05:
At least the module is correctly loading now. Could you post the output of "sudo rfkill list all" ?

Sahib911, 2020-06-05:
chronos@localhost / $ sudo rfkill list all
0: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no

Sahib911, 2020-06-06:
@sebanc hey , in the unstable build you shared youtube (website) is not working .Or any other video on websites . 
And are there any updates on Bluetooth . 
Plus my touchpad is very sensitive 
And if you need anything to be tested let me know .

sebanc, 2020-06-06:
Let's focus on bluetooth first. I cannot really make sense of why the module is correctly loaded but no bluetooth interface is shown...

What's the output of "sudo hciconfig -a" ?

Sahib911, 2020-06-06:
"sudo hciconfig -a" Does nothing 

crosh> shell
chronos@localhost / $ sudo hciconfig -a
Password: 
chronos@localhost / $

sebanc, 2020-06-06:
Ok, I think I got it, I did not see there was a userspace part also...

Could you download the rtbt binary for here and copy it in /usr/bin:
https://github.com/loimu/rtbth-dkms/blob/master/tools/rtbt

Then could you run the below command in crosh shell:
`sudo mknod /dev/rtbth c 192 0` /usr/bin/rtbt &`

Sahib911, 2020-06-06:
How can i copy it to /usr/bin ?

sebanc, 2020-06-06:
"sudo cp ~/Downloads/rtbt /usr/bin/"

Sahib911, 2020-06-06:
chronos@localhost / $ sudo mknod /dev/rtbth c 192 0
Password: 
mknod: /dev/rtbth: File exists
chronos@localhost / $ sudo /usr/bin/rtbt &
[1] 4164
chronos@localhost / $ sudo: /usr/bin/rtbt: command not found

sebanc, 2020-06-06:
`/usr/bin/rtbt: command not found` means that the rtbt binary has not been properly copied to /usr/bin/ folder.

Did "sudo cp ~/Downloads/rtbt /usr/bin/" return an error ?

sebanc, 2020-06-06:
Ah maybe you need to run "sudo chmod 0755 /usr/bin/rtbt"

Sahib911, 2020-06-06:
chronos@localhost / $ sudo cp ~/Downloads/rtbt /usr/bin
Password: 
chronos@localhost / $ sudo mknod /dev/rtbth c 192 0
chronos@localhost / $ sudo chmod 0755 /usr/bin/rtbt
chronos@localhost / $

Sahib911, 2020-06-06:
hronos@localhost / $ sudo chmod 0755 /usr/bin/rtbt
Password: 
chronos@localhost / $ sudo mknod /dev/rtbth c 192 0
mknod: /dev/rtbth: File exists
chronos@localhost / $ sudo /usr/bin/rtbt &
[1] 8017
chronos@localhost / $ Open dev fail: No such device or address

sebanc, 2020-06-06:
Could you reboot, run the following commands and post a dmesg ?
```

Sahib911, 2020-06-06:
[dmesg.txt](https://github.com/sebanc/brunch/files/4739945/dmesg.txt)
Here

sebanc, 2020-06-06:
The module is not loading as per the first issue we solved (see below), could you re-perform this first, then run the 2 commands and post a dmesg ?

> Could you try to add "modprobe bluetooth" before "insmod /lib/modules/rtbth.ko" in /etc/init/rtbth.conf ?
>
> From:
>
> ```
> start on stopped udev-trigger
>
> script
> 	insmod /lib/modules/rtbth.ko
> end script [...]

Sahib911, 2020-06-06:
chronos@localhost / $ sudo mknod /dev/rtbth c 192 0
mknod: /dev/rtbth: File exists
chronos@localhost / $ sudo /usr/bin/rtbt &
[2] 3843
chronos@localhost / $ init complete

It worked . 
[dmesg.txt](https://github.com/sebanc/brunch/files/4739990/dmesg.txt)
Thanks a lot man.

sebanc, 2020-06-06:
Last thing, could you try to replace the content of /etc/init/rtbth.conf with the below text ?
```
start on stopped udev-trigger

script
	modprobe bluetooth
	insmod /lib/modules/rtbth.ko
	rtbt
end script
```
Bluetooth should work out of the box at startup without additional commands.

sebanc, 2020-06-06:
The youtube issue is most likely related to a sound patch I will probably revert in the next build but I need to receive feedback from another issue first.

For the touchpad, you can try "options=alt_touchpad_config" or to modify manually the first section of /etc/gestures/40-touchpad-cmt.conf with options you will find in the other sections of the same file.

Sahib911, 2020-06-06:
okay ,
 alt touchpad config doesn't work for me i'll try the other method.
Bluetooth is working fine if it's open in the chrome's setting or when it's looking for devices .is there a solution for that .

sebanc, 2020-06-06:
> Bluetooth is not working out of the box , i have to use those 2 commands everytime . I have changed the content of config file.

could you try to replace the content of /etc/init/rtbth.conf with the below text:
```
start on stopped udev-trigger

script
	modprobe bluetooth
	insmod /lib/modules/rtbth.ko
	mknod /dev/rtbth c 192 0
	rtbt
end script
```

> Bluetooth is working fine if it's open in the chrome's setting or when it's looking for devices .is there a solution for that .

Sorry but I don't understand what you mean

Sahib911, 2020-06-06:
> could you try to replace the content of /etc/init/rtbth.conf with the below text:
>
> ```
> start on stopped udev-trigger
>
> script
> 	modprobe bluetooth
> 	insmod /lib/modules/rtbth.ko
> 	mknod /dev/rtbth c 192 [...]

Still no luck , now those 2 commands are also not working
[dmesg.txt](https://github.com/sebanc/brunch/files/4740771/dmesg.txt)

Sahib911, 2020-06-06:
> > Bluetooth is working fine if it's open in the chrome's setting or when it's looking for devices .is there a solution for that .
> 
> Sorry but I don't understand what you mean

I meant , that while listening to music the audio is stuttering and is fixed only when in bluetooth settings page or when devices are being scaned  (chrome://bluetooth-internals/#devices),

Sahib911, 2020-06-15:
works in the new build . Thank you very much and keep up the good work .

closed 2020-06-15 by Sahib911

ayushbathrey, 2020-08-01:
Not working in latest build.
When I try to connect to my tv it pairs but does not connect, and my phone won't appear while scanning.
[dmesg.txt](https://github.com/sebanc/brunch/files/5010153/dmesg.txt)