sebanc/brunch #253: Installation died after updating.

OkameProductions, 2020-06-02:
After attempting an update (with `options=enable_updates`), it seemed to go fine.
Brunch detected an update, applied and rebuilt rootfs, and then the install died.
Running Brunch r83.

The last message sent into the console was
```
sh: ext2: unknown operand
```
before crashing and restarting.
`traps` also complained about an invalid opcode.

Reinstalling was easy, since dualboot, but I'm a little sad that the update couldn't go through and that I have to reconfigure ChromeOS again.
The project is really impressive, keep it up! I used to use Project Croissant, but it fell out of favor due to lack of maintenance. This project is the next best thing, and it's very promising!

sebanc, 2020-06-02:
Google has included in r83 cpu instructions which were not previously enabled in many recovery images, with your cpu you have to use samus recovery image to update to r83.

OkameProductions, 2020-06-02:
Ah, darn it.
Do you think I could reapply samus' recovery image with `sudo chromeos-update`?

sebanc, 2020-06-02:
yes "sudo chromeos-update -r <path_to_unzipped_samus_recovery>" should work (backup your important data just in case)

OkameProductions, 2020-06-02:
I just reinstalled the img, so I don't have anything on it yet.
I'll download the image and attempt to apply it.
Thanks so much!

OkameProductions, 2020-06-02:
Closing for now, if anything happens I'm going to reopen.

closed 2020-06-02 by OkameProductions