sebanc/brunch #254: USB Installer not booting

raagk, 2020-06-03:
Hey, thanks for your work !
I made bootable usb in linux as mentioned in guide, used **Rammus recovery**.
when I select use for boot installer get's to the point where it is copying files stays there , and without completing reboots and boots in windows (attempting dual boot).
I have a 16 GB pendrive.
Also I had successfully booted once but after setup and a reboot is was stuck on black screen with cursor, so I made usb installer again in linux but no luck.
Hope my issue will be solved.
Thanks !

Specs : Intel i3 5010U (5th generation)
             8 GB DDR3 RAM
             No external gfx

sebanc, 2020-06-03:
Hi,

Could you post a short video of the boot process ?

raagk, 2020-06-03:
Sorry but it was problem of my USB Drive, I used another and it worked flawlessly !
Also just want to ask can chrome os boot without usb ? installed grub on hard drive but I don't know how to add entry on grub installed on hdd.

closed 2020-06-03 by raagk

sebanc, 2020-06-03:
Are you dual booting from widows or linux ? How did you install grub ?

raagk, 2020-06-03:
I installed linux + windows +ChromeOS but then I deleted partition linux was installed on but it still shows OS boot Manager (UEFI) -  ubuntu (hard disk name) and booting through it displays  GRUB with minimal bash editing window like this : https://www.tecmint.com/wp-content/uploads/2017/09/Ubuntu-Grub-Console.png

sebanc, 2020-06-03:
Ok what you will need to do:
- Identify and mount your hdd efi partition in chromeos.
- Copy the content of /efi/boot/ on chromeos partition 12 to your hdd efi partition in /efi/ubuntu/ folder.
- Replace /efi/ubuntu/grub.cfg content with the dual boot script.

raagk, 2020-06-03:
Done ! I had to copy whole efi/boot/ and brunch.dr from USB to EFI/ubuntu after cleaning the folder ,it boots successfully without USB.
 But problem is PlayStore is not working!

sebanc, 2020-06-03:
Could you try deleting google sync ?
https://support.google.com/chromebook/answer/2914794?hl=en

raagk, 2020-06-04:
Hey ! thanks9 for help but it did't work : (