sebanc/brunch #257: Cant alt-tab or windows or whatever

efbe2121, 2020-06-03:
I've updated to your newest brunch release, and my Chrome OS is at 83. I noticed something peculiar after i updated. Plant vs Zombies 2, that usually works pretty normal, now i cant alt-tab it to change windows. Not even that, if i press windows button, it does nothing. I only tested it on 2 apps, Get Rich, and PvZ 2. So is there something wrong? I only enable the updates option on my grub

efbe2121, 2020-06-03:
to be more precise, it similar like the android apps are locking my view, locking the keyboard. I cant even use the on-screen keyboard. It shows up like milisecond, then disappear

PixelOFC, 2020-06-03:
try changing you keyboard layout

efbe2121, 2020-06-12:
i dont think keyboard layout is the problem here, cuz i can use alt-tab and windows button normally outside of this android apps

sebanc, 2020-06-13:
Hi,

I doubt this change of behavior is brunch related. The only thing that comes to my mind which might have this kind of impact would maybe be the line "--enable-hardware-overlays="single-fullscreen,single-on-top" in /etc/chrome_dev.conf.

Could you try running "sudo nano /etc/chrome_dev.conf", removing that line, rebooting and checking if it changes something ?

efbe2121, 2020-06-13:
Ah i've downgrade my brunch version to stable version. I will try to update it again and see if the problem occurs

efbe2121, 2020-06-13:
[image]

efbe2121, 2020-06-13:
And no, the problem still exist. I cant use windows button or even alt-tab

sebanc, 2020-06-13:
Indeed it seems weird, before I look further into it, could you confirm if you have the same issue with the latest test build:
https://github.com/sebanc/brunch-testing/releases

(You need to accept an invite email received from github first)

efbe2121, 2020-06-13:
I will try it and inform you later since ive reinstalled the chrome OS itself with your stable release. This issue indeed weird and happened after using it for a while, like 2 or 3days or so on

efbe2121, 2020-06-15:
Hi, i've installed the latest test build. And the issue seems gone.
But to make sure, im going to test with this brunch for few days just to make sure
Especially i dont know what triggers the issue. Probably the sleep when i close the laptop lid

efbe2121, 2020-07-23:
After a while ive observed, this issue is gone :). The latest release somehow fix this issue. Thanks for your hardwork!

closed 2020-07-23 by efbe2121