sebanc/brunch #258: Would you mind adding a driver for the 2016+ MacBook Pros (Links)

DiegoMagdaleno, 2020-06-03:
Hi, I am a ChromeOS enthusiast, and I really like your work, I use it on my MacBook Pro, and most of it works, except for the keyboard (sound doesn't but I don't know if you would bother adding that too), kernel modules have been created to run this, and I would like to see if you would bother adding them, I tried on my own but if I am honest I didn't get anywhere.
 
Thank you for your amazing work :).

Here are the links:

- [Keyboard and Touchpad Driver](https://github.com/roadrunner2/macbook12-spi-driver)

And if you wish:

- [Sound driver](https://github.com/davidjo/snd_hda_macbookpro)

Thanks a lot again. Greetings!

sebanc, 2020-06-04:
Hi,

I added the apple spi keyboard and touchpad driver in the unstable build available here:
https://github.com/sebanc/brunch-testing/releases

Could you try it ?

For sound, the state of the module is too experimental and I don't want to risk breaking people's speakers... However, you can build the module directly in brunch if you want to add it:
- Download chromebrew
- Clone the sound driver git
- Run make
- Replace the module with the same name in "/lib/modules/4.19..../kernel/sound/...." with your built one.

DiegoMagdaleno, 2020-06-04:
Thanks a lot, and sure I will do, thanks a lot for your work :), I report back if the sound is working in a bit!

DiegoMagdaleno, 2020-06-04:
Okay so I downloaded the new one, i just have to wait now for the download of the ChromeOS recovery, do the process and then I notify you.

DiegoMagdaleno, 2020-06-04:
There is only 1 issue:

- Moving with the trackpad doesnt work, clicking does but you cant move the cursor, also two finger right click doesnt work, in Linux it does so I don't know if this is an issue with the driver or chromeOS

sebanc, 2020-06-04:
Try with "options=alt_touchpad_config"

DiegoMagdaleno, 2020-06-04:
I try the touchpad options later, for now, sound isnt working with the driver, i dont know if I did this the right way or not.

[image]

sebanc, 2020-06-04:
No, actually I did not provide the complete path. Run "find /lib/modules | grep "cirrus" to identify the path of the original module and then replace the existing file with yours.

DiegoMagdaleno, 2020-06-04:
Okay so I can finally report back and here are the results:

- Trackpad only works for clicking you cant move with the trackpad 
- Sound isnt working

The rest is working flawlessly and the work around for hibernation on Linux works on ChromeOS too, i trust that in the future support for this MacBooks will be getting better in Linux, if you don't mind I would find very very useful to have a git history when you added the MacBook pro keyboard driver, so in the future I can contribute on my own, thanks a lot for your amazing work, greetings!

sebanc, 2020-06-04:
Could you attach a dmesg ? (run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and attach dmesg.txt from your downloads folder).

DiegoMagdaleno, 2020-06-04:
Sure here you go
[dmesg.txt](https://github.com/sebanc/brunch/files/4732091/dmesg.txt)

sebanc, 2020-06-04:
The sound module does not seem to be probed, could you try to reboot, then press CTRL+ALT+F2, login as "root" and run "insmod <path_to_the_sound_module>", then CTRL+ALT+F1, login, test sound and attach another dmesg.

DiegoMagdaleno, 2020-06-04:
Looks like it can’t load the module because it contains an unknown symbol 

PS: Sorry for the Photo in my iPad I only have GitHub there

[image]

sebanc, 2020-06-04:
Thanks for the photo, could you build the module again and post the output of the "make" command here ? I think something went wrong at build time.

DiegoMagdaleno, 2020-06-04:
No problem and yes sure, I could, and also I ran scrip+evtest in the Apple SPI Touchpad here is it, hope some of it can helps, it does detect when it is being touched. And sure I show the modules in a bit. 

And here is the evtest on the touchpad:
[chromeOSOnMacbookDebugTouchpad.txt](https://github.com/sebanc/brunch/files/4732449/chromeOSOnMacbookDebugTouchpad.txt)

DiegoMagdaleno, 2020-06-04:
And here is the output of the sound driver while being built. 
[buildingCirrusDriverIssue.txt](https://github.com/sebanc/brunch/files/4732454/buildingCirrusDriverIssue.txt)

DiegoMagdaleno, 2020-06-04:
The Apple Magic Trackpad works and i might have an idea of modifying the configuration.

[Got the idea from here ](https://neverware.zendesk.com/hc/en-us/community/posts/360000764527-Neverware-MacBook-Air-Touchpad-Driver)

I report back in a bit.

DiegoMagdaleno, 2020-06-04:
Okay so I reconfigured the trackpad and that didnt see to help, the trackpad is getting detected however and tapping on it does register an event I think, probably is some configuration issue on ChromeOS?

DiegoMagdaleno, 2020-06-05:
@sebanc So a little update on some stuff:

- Touchpad isnt working, however it is properly detected, this might be a configuration issue of ChromeOS itself.

- Sound undefined symbol is because this is a GNULIBC exclusive symbol, what does this mean? This means only GNU operating systems can use this, and ofc we know ChromeOS doesnt use GNULIBC, however a quick fix is this code:

```c
unsigned long __stack_chk_guard;
void __stack_chk_guard_setup(void)
{
     __stack_chk_guard = 0xBAAAAAAD;
}

void __stack_chk_fail(void)                         
{
 /* Error message */                                 
}
```
insmod doesnt fail anymore, however there is still no sound, this might be a configuration issue of ChromeOS at the end of the day.

sebanc, 2020-06-06:
I think the correct way of fixing the "stack_chk_guard" issue would be to add
`CFLAGS_snd-hda-codec-cirrus.o := -fno-stack-protector`

after the below lines in the Makefile:
```
snd-hda-codec-cirrus-objs :=	patch_cirrus.o
obj-$(CONFIG_SND_HDA_CODEC_CIRRUS) += snd-hda-codec-cirrus.o
```

Could try this build method, replace the kernel module and post another dmesg ?

sebanc, 2020-06-08:
I might have broken something related to sound in the previous brunch testing release, could you try to build the sound module with the above fix using the new test release:
https://github.com/sebanc/brunch-testing/releases

DiegoMagdaleno, 2020-06-08:
Yeah I will test this today and notify any progress, sorry for not replying before GitHub wasn’t showing the notification

sebanc, 2020-06-12:
@canayaf713 You should now have access to the testing repo. This feature will also be included in the next standard release which should happen during the weekend.

canayaf713, 2020-06-12:
Thanks! My keyboard is working now, however I am having the same trackpad issue. I will try to figure out what's causing the problem and update if there is any progress.

DiegoMagdaleno, 2020-06-16:
@canayaf713 for your trackpad could you try https://neverware.zendesk.com/hc/en-us/community/posts/360000764527-Neverware-MacBook-Air-Touchpad-Driver 

Trying something like that, click works, but not scrolling for reason.

DiegoMagdaleno, 2020-06-16:
A possible fix will be to use libinput, since that works on Linux, but I dont know if this is possible, I know cloudready does that.

sebanc, 2020-06-16:
Unfortunately, cmt is the equivalent for libinput in chromeos and I don't think I can replace it.

canayaf713, 2020-06-23:
@DiegoMagdaleno Does your trackpad work? I tried the link you provided but only scrolling and clicking works.

DiegoMagdaleno, 2020-06-23:
@canayaf713 it doesnt however, "Scrolling" is more progress than I ever got, continue playing with the configuration and you might get it.

Joshagos, 2021-09-17:
WiFi still isn't working on my 2017 MacBook Air.