sebanc/brunch #260: Cannot use the microphone and camera on Dell Inspiron 5379

JakunGamer, 2020-06-04:
So I needed camera and microphone for Microsoft Teams. However, the camera is only show some sort of black and white photo. I guess it was using the Windows Hello Camera, not the main camera which is not good. The microphone also doesn't work. 
I'm using Dell Inspiron 5379 with Brunch v83 and ChromeOS Nami v83. Didn't use rammus cause I felt that Nami is more stable.

sebanc, 2020-06-04:
Hi,

Could you try mounting partition 7 of chromeos and replacing /patches/99-camera.patch file with the one in the below zip file ?
[99-camera.sh.zip](https://github.com/sebanc/brunch/files/4727657/99-camera.sh.zip)

Then reboot, wait for rootfs and rebuild and try camera again.

JakunGamer, 2020-06-04:
> Hi,
> 
> Could you try mounting partition 7 of chromeos and replacing /patches/99-camera.sh file with the one in the below zip file ?
> [99-camera.sh.zip](https://github.com/sebanc/brunch/files/4727657/99-camera.sh.zip)
> 
> Then reboot, wait for rootfs to rebuild and try camera again.

One question: Which one is partition 7: /dev/sda7 or /dev/loop7 or something else?

sebanc, 2020-06-04:
If you are using single boot, it is /dev/<your_device>7.

If you are using dual boot, it would be /dev/loop15p7

JakunGamer, 2020-06-04:
> If you are using single boot, it is /dev/<your_device>7.
> 
> If you are using dual boot, it would be /dev/loop15p7

Well I've done what you told me to do but it didn't work.

sebanc, 2020-06-29:
Sorry forgot to update this issue but if you are still looking for a solution the camera will should work with "reef" image as it has the old but more compatible chromeos usb camera framework.

JakunGamer, 2020-07-05:
> Sorry forgot to update this issue but if you are still looking for a solution the camera will should work with "reef" image as it has the old but more compatible chromeos usb camera framework.

So what about the microphone :)

sebanc, 2020-07-05:
The microphone is still an unresolved issue, there are some workarounds avilable in the issue #56.

JakunGamer, 2020-07-06:
> The microphone is still an unresolved issue, there are some workarounds available in the issue #56.

Thanks man, hope this will get fix soon