sebanc/brunch #261: can i install the image grow?

djgerman10, 2020-06-04:
It turns out that laptops with this image have the same specifications as mine. I was wondering if I can install this image (grow).

specifications of my laptop [https://www.intel.com/content/www/us/en/support/articles/000006284/education/intel-education-hardware.html](https://www.intel.com/content/www/us/en/support/articles/000006284/education/intel-education-hardware.html)

sebanc, 2020-06-04:
Hi,

Aside from the kernel, all chromebooks recovery images are very similar. You can try enguarde image but as brunch replaces the kernel with its own I doubt it will provide any advantage over using rammus recovery image.

djgerman10, 2020-06-04:
Another query would be if I have the image in version 80 I have to also put the brunch of version 80 right?

because I put vercion 83 and it is in chrome os coming soon

remains on the screen chrome os coming soon in uefi and legacy

sebanc, 2020-06-04:
> Another query would be if I have the image in version 80 I have to also put the brunch of version 80 right?

Not necessarily brunch r83 is backwards compatible with r80 recovery images.

> 
> because I put vercion 83 and it is in chrome os coming soon
> 
> remains on the screen chrome os coming soon in uefi and legacy

I don't understand what you mean, I have never seen a "chrome os coming soon" message, could you post a short video of the boot process ?

djgerman10, 2020-06-04:
[https://www.youtube.com/watch?v=phA-pM9qCis](https://www.youtube.com/watch?v=phA-pM9qCis)
5:55
remains on the screen chrome os coming soon in uefi and legacy

sebanc, 2020-06-04:
samus

djgerman10, 2020-06-04:
### apparently the page fell
Do you know any alternative link to download the image of "samus"?

[image]

_Thank you very much, sorry for the inconvenience._

sebanc, 2020-06-04:
Try:
https://webcache.googleusercontent.com/search?q=cache:js20L4Y_5LYJ:https://cros-updates-serving.appspot.com/+&cd=1&hl=fr&ct=clnk&gl=fr

djgerman10, 2020-06-05:
install "enguarde" image with brunch r80, apparently it has 83 incompatibility with "enguarde" image.

I can't find where to activate play store

[image]

sebanc, 2020-06-05:
enguarde image does not have android apps support:
https://www.chromium.org/chromium-os/chrome-os-systems-supporting-android-apps

Not sure why you are trying that hard to install this image...

djgerman10, 2020-06-05:
Now I understand, I thought that this image is more optimized for laptops like mine, but if it does not have android support it is not worth it.

gravare image "samus"

djgerman10, 2020-06-05:
another query chrome os is compatible with the graphics intel gma 3600?
because fydeos and cloudready is not, I was wondering if official chrome is?

_and again I appreciate that you share your knowledge, thank you very much_

sebanc, 2020-06-05:
I don't think I have seen reports on this hardware but I have seen chromeos run on weirder configs. I cannot guarantee anything but it's probably worth trying it.

closed 2020-06-26 by sebanc