sebanc/brunch #263: HP Spectre X2 (2015,skylake) support

daemongloom, 2020-06-04:
HP Spectre X2 12-a000 series.
CPU: Intel® Core™ M5-6Y30

- [x] Does the Android Subsystem (Android apps) run?
- [x] Is Linux container install option available?
- [x] Does Linux container install?
- [x] Does Linux terminal run?
- [x] Is Bluetooth working?
- [x] Is WiFi working?
- [ ] Is the camera working?
- [ ] Is microphone working?
- [x] Does trackpad and gestures work?

Volume and power buttons require intel-hid patch. I've attached it for r83 branch. [0001-Fix-power-and-volume-buttons-on-HP-Spectre-X2.txt](https://github.com/sebanc/brunch/files/4729517/0001-Fix-power-and-volume-buttons-on-HP-Spectre-X2.txt)
One of back cameras works (maximum available resolution is 2.1MP, should be 8MP), front one doesn't. Back camera is shown as front-facing in default Camera app.
Suspend leaves keyboard backlight on in both usual and suspend_s3 modes.
Switching to tablet mode (via detaching keyboard or folding it to the back of tablet) works, automatic rotation works.
I will try to fix other issues but it would be great if you include patches in next builds.

It should be noted that 2015 version of tablet differs a lot from previous one (https://github.com/sebanc/brunch/issues/186)

sebanc, 2020-06-04:
Hi,

I added the button patch and attempted a fix for microphone in the below build:
https://github.com/sebanc/brunch-testing/releases

Could you try it ?

daemongloom, 2020-06-05:
Buttons and microphone works ok in that build.

sebanc, 2020-06-05:
Buttons working was expected but microphone is a good surprise, could you please attach a dmesg ? (run "dmesg > ~/Downloads/dmesg.txt" and attach dmesg.txt from your downloads folder)

Some devices have camera issues with "rammus" recovery image that I have not yet been able to fix, they should work correctly with "samus" recovery image.

daemongloom, 2020-06-05:
I've attached logs and hardware info.
[lsusb_v.txt](https://github.com/sebanc/brunch/files/4734084/lsusb_v.txt)
[lspci_v.txt](https://github.com/sebanc/brunch/files/4734085/lspci_v.txt)
[lsmod.txt](https://github.com/sebanc/brunch/files/4734086/lsmod.txt)
[dmesg.txt](https://github.com/sebanc/brunch/files/4734087/dmesg.txt)

sebanc, 2020-06-05:
Thanks for all the logs :)

daemongloom, 2020-06-18:
Hm. I have strange issue with tablet mode. Sometimes it works perfectly, disconnecting keyboard (or folding it back) changes interface on-screen keyboard popup automatically and automatic rotation works. But sometimes it just ignores everything. No on-screen keyboard, no interface change and no rotation. I've attached both dmesg logs.
[log_no_tablet_mode.log](https://github.com/sebanc/brunch/files/4799702/log_no_tablet_mode.log)
[log_with_tablet_mode.log](https://github.com/sebanc/brunch/files/4799703/log_with_tablet_mode.log)

sebanc, 2020-06-20:
I see nothing weird in the logs, also the driver used to report tablet mode is very basic and is the same as my Surface Go where it works fine. This issue might be caused by the bios/ec not sending the events properly, a bios update might help if you are not on the latest one.

daemongloom, 2020-07-13:
Ok, even stranger things happen now. When I fold keyboard backwards (or disconnect it) interface changes, but on-screen keyboard does not popup automatically and automatic orientation is not enabled.