sebanc/brunch #264: Booting with Clover / Grub

raagk, 2020-06-04:
Hey!
I have clover boot loader installed on my PC to boot OSX . However i want to boot Chrome OS.
So is there any way to boot clover from grub or grub from clover ?
Please answer

raagk, 2020-06-04:
Can you send link Please !

sebanc, 2020-06-04:
Here: https://t.me/chromeosforpc

raagk, 2020-06-04:
Thanks , also I wanted to ask for legacy boot , where should I apply -l in command for installation on USB? I applied it after chromeos-install.sh and it didn't work , my PC didn't boot on Pendrive.

closed 2020-06-04 by raagk

sebanc, 2020-06-04:
"-l" option is legacy efi.

What you are looking for is probably the "mbr_support.tar.gz" package in the master branch of this repo. Extract brunch release, then "mbr_support.tar.gz" in the same directory and proceed as usual.

raagk, 2020-06-05:
Hey ! It worked !
Also is there any grub entry to disable Nvdia GPU?