sebanc/brunch #267: Installation fails on Atom N450 / ASUS Eee PC

Jet68, 2020-06-04:
For some days I am searching for a way to get Chrome OS runnning on this old Netbook. With your work I am closer than ever. After solving the issues to get a USB stick prepared with a bootable image (Bios, no UEFI - MBR no GPT) the installation of Chrome OS starts but after some time stops in a line with
random: crng init done
There it stays for some time and when it continues the text is rushing on until some lines with 
**SELINUX**  are visible only for a split second until the netbooks restarts and repeats the same (booting into grub and starting the Chrome OS install)

Any idea what can be done?
Has anyone been able to run Chorme OS on such a device?

Any help is appreciated.

sebanc, 2020-06-04:
Hi,

Your issue and cpu looks very similar to issue #225, unfortunately there are no current chromeos recovery images compatible with your cpu.

If you are looking for android apps support, FydeOS or "PrimeOS Classic" might be good alternatives.

Jet68, 2020-06-04:
Yes, seems to be a similar issue. Have not checked the closed issues.Thank you for your work and the fast answer.Keep it upVon meinem Samsung Galaxy Smartphone gesendet.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/sebanc/brunch/issues/267#issuecomment-639104880",
"url": "https://github.com/sebanc/brunch/issues/267#issuecomment-639104880",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

pxsepulv, 2020-06-15:
FydeOS doesn't work on this CPU (at least in my case). Try using Android from https://www.android-x86.org/ for now.

Jet68, 2020-06-16:
> FydeOS doesn't work on this CPU (at least in my case). Try using Android from https://www.android-x86.org/ for now.

I confirm this. Did not work for me as well. Got Cloudready running which is only half of what I was looking for. May give android86 a try as well. 
Thank you for your answer.

pxsepulv, 2020-06-16:
I'm trying FydeOS in another PC and it's not the great thing after all. I tried Androidx86 8 in an Atom Netbook and works great (Android 9 doesn't work on it). Download Sentio Desktop app for a more Desktop like environment. It seems I'll do this after all.

closed 2020-06-26 by sebanc