sebanc/brunch #269: Video and Wifi issues

raytrix71, 2020-06-05:
Hello, I'm currently using brunch chrome os on HP laptop 15db0097nf, I have an issue with wifi, the only way to connect to internet is with a cable. My computer support rtf8723de combo wifi and bluetooth. I tried to add with this method  "shell" then "sudo edit-grub-config" adding a the end of the line "options=rtl8723de" and save, doing the reboot nothing works, I made sure than what I written is saved ans that's the case. I don't know what I can do.

I experience an issue with YouTube video playing on chrome.

sebanc, 2020-06-05:
Hi,

Could you attach a dmesg to this issue ? (run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and attach dmesg.txt from your downloads folder to this issue)

raytrix71, 2020-06-05:
thanks for your reply
[dmesg.txt](https://github.com/sebanc/brunch/files/4736753/dmesg.txt)

sebanc, 2020-06-05:
Sorry I forgot to mention but I would need a dmesg taken just after a reboot.

Also, which brunch versionand which recovery image do you use ?

raytrix71, 2020-06-05:
its done 
[dmesg (1).txt](https://github.com/sebanc/brunch/files/4736896/dmesg.1.txt)

raytrix71, 2020-06-05:
I use grunt recovery 80 and  Brunch r80 k4.19 stable 20200407

raytrix71, 2020-06-05:
maybe should i use 81?

sebanc, 2020-06-05:
Ok, I cannot see much in the dmesg because the failing graphics are taking all the logs. Anyway, I think the rtf8723de adapter only started being supported in the r83 testing version and it even though it is a test version it appears to be quite stable. Could you try this one ?

raytrix71, 2020-06-05:
So I installed the last version of brunch, the WiFi worked for two minutes and now it’s over

sebanc, 2020-06-06:
I have disabled wifi powersave in the latest brunch testing repo release, could you try it ?
https://github.com/sebanc/brunch-testing/releases

raytrix71, 2020-06-06:
Hi it’s me again , I’ve done the firmware update , everything was working well for 10 to 20 minutes , and then I’ve face a problem with the WiFi , I got disconnected. The only way to fix it is to restart the computer. Please help me

subash-uduwaka, 2020-06-21:
WIFI is not working in Dell Inspiron 3543. First boot there wasn't a wifi logo. After adding "options=iwlwifi_backport" the logo is appeared but not showing networks

raytrix71, 2020-06-22:
Have tried other WiFi option like broadcom?

sebanc, 2020-06-23:
@raytrix71 I tried to improve rtl8723de support in latest test release, could you try it ? (without "options=rtl8723de")
https://github.com/sebanc/brunch-testing/releases

subash-uduwaka, 2020-06-23:
> @raytrix71 I tried to improve rtl8723de support in latest test release, could you try it ? (without "options=rtl8723de")
> https://github.com/sebanc/brunch-testing/releases

Sorry this link is not working! 404 error is showing. Plz give me a updated link!

sebanc, 2020-06-23:
@subash-uduwaka It would not help in your case, you do not have the same wireless adapter