sebanc/brunch #270: Issue using DisplayLink Dock

ruhess, 2020-06-05:
Hello

Thank you for all the good work. 
I experience ab issue with a DisplayLink dock

Everytime I connect a Fujitsu PR08 port replicator the system  reboots and gets stuck in a reboot loop until I disconnect the dock

I use brunch_r81.k4_19_stable_20200510 with chromeos_12739.111.0_fizz_recovery_stable-channel_mp-v12 on an NUC7i5

During the reboot it correctly detects the PR08 but crashes once the login screen shows up
[dmesg.txt](https://github.com/sebanc/brunch/files/4737759/dmesg.txt)

sebanc, 2020-06-06:
Hi,

Could you try the latest release from the "brunch-testing" repo ?
https://github.com/sebanc/brunch-testing/releases

ruhess, 2020-06-06:
Thank you very much for your support.
I tried bruch_r83 but unfortunately no difference.
I noticed that the system only crashes when a monitor is connected to the docking station. Without any monitor the USB ports of the docking work.
Since I don't know how to access the crash log the attached dmesg.txt shows a boot-up with the docking station connected but no monintor connecetd to to docking station. Not sure if this is helpful
[dmesg.txt](https://github.com/sebanc/brunch/files/4739871/dmesg.txt)

pfroo40, 2020-06-24:
I am also having this issue on the latest r83 brunch kernel and rammus recovery, I have tested with two different Displaylink devices and on USB-C and A. I have a UMPC 2 in 1 with an Intel N5000 Pentium (Topjoy Falcon 8).

sebanc, 2020-06-24:
Unfortunately, I have no idea what could cause that. To go further with this issue, I would need a short video of the brunch boot text when a display is connected to the adapter until the bootloop occurs.

sebanc, 2020-06-25:
I tried something in the latest unstable available below:
https://github.com/sebanc/brunch-testing/releases

Could you try it and see if it improves things ?

pfroo40, 2020-06-25:
I installed the testing version Brunch r83 k4.19 unstable 20200625
successfully and I tested successfully as well. Thank you!! I'll test on my
other docks and let you know if I have any issues, but I'd expect they will
work fine.

One of my devices is USB powered (portable Asus monitor) and it doesn't
seem to get enough power to stay on when connected to my USB-A 3.0, which I
don't really care about, since USB-C works fine (as well as through a USB-C
hub I have with external power passthrough). It blinks an image and then
restarts. So, not really a big deal, unless there's a reason you may want
to know that the power output of the USB-A port seems lower than through
Win10.

One question if you have time, I'm trying to work around the kernel missing
the driver for the rotation sensor and 360 hinge open sensor for tablet
mode, I've followed your instructions to add the kernel option to enable
tablet mode, but I'm wondering if there is a shell command to disable it
again (instead of rebooting)? I tried the same for clamshell_mode but
doesn't seem to do anything, but I'm new to ChromeOS so that was just a
guess. Since there's no toggle, at least in shell it'd be cool if I could
use commands and bind them to a shorter shortcut text string in shell.

Anyway, I know tablet mode is YMMV if not working out of box, so not a big
deal.

Thanks again

sebanc, 2020-06-25:
Nice to know that the issue appears to be mostly fixed :)

The command to disable tablet mode is:
`echo 0 | sudo tee /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode`

pfroo40, 2020-06-26:
Perfect, thanks for that, too!

daemongloom, 2020-06-27:
I can confirm that latest unstable version works with my hp displaylink dock. The only issue is that tablet's touchscreen coordinates are totally wrong.

sebanc, 2020-06-27:
What do you mean by tablet's touchscreen coordinates ? Do you mean your device touchscreen is not correctly reporting input once the external monitor is added ? or is it an external tablet device ?

daemongloom, 2020-06-27:
> Do you mean your device touchscreen is not correctly reporting input once the external monitor is added ?
This. My device is HP Spectre X2 tablet.

sebanc, 2020-06-29:
For now, I have no idea why the touchscreen mapping is incorrect when an external monitor is present but the issue #319 seems to be specifically related to this problem so I will perform the follow up there.

I am closing this one has the displaylink part seems fixed.

closed 2020-06-29 by sebanc