sebanc/brunch #271: Play Store doesn't load on USB drive for 2nd Generation Intel chip PC

dshnrob, 2020-06-05:
Hi,

First of all, great job with the Brunch/Chrome OS.  I am loving how well it responds off of my USB drive.  But I can't get Google Play store to load.  It tries, fails and then asks me to try again.

Here's my setup:  I am using the Samus Recovery image 80 as well as the accompanying 80 Brunch   config.  I have already turned off the Google Cloud recovery/sync to no avail.  

Is there anything I can do to get Play Store running?  I am hoping to hear others who have had success getting their Intel Sandy Bridge processors working with this Chrome OS.

Thanks!

Sahib911, 2020-06-06:
Try enabling virtualization in BIOS  .

sebanc, 2020-06-06:
Hi,

Could you try deleting google sync ?
https://support.google.com/chromebook/answer/2914794?hl=en

Also brunch r83 testing is backwards compatible with r80 and appears to be stable, could you update to it ?

khanhtranngoccva, 2020-06-06:
After that, wait from 1 minute to 1 minute and a half until android is ready to load the data. If you want to check when it's ready, press Ctrl+Alt+F2, type chronos and issue command "dmesg". Make sure you have this line spawned before logging in: [image]

djgerman10, 2020-06-06:
install the image "reks" that works better for intel 2nd gen

download image r81 and brunch r83

francarl, 2020-06-06:
> After that, wait from 1 minute to 1 minute and a half until android is ready to load the data. If you want to check when it's ready, press Ctrl+Alt+F2, type chronos and issue command "dmesg". Make sure you have all the lines in the red rectangle spawned before logging in: [image]

Thank you for the tip, it works. Waited to login until `dsmeg | grep capability` on shell gave result.

dshnrob, 2020-06-07:
Hi there,

Thanks for all the suggestions.  Unfortunately, I haven't had any success yet getting the Google Play Store to load.

This is what I have tried so far:
I used the brunch r83 with the samus r80 and r81.  Both fail on initial configuration of the Google Play Store.  I also turned off Google Sync on both with no success.

I also, per suggestion, tried the reks image r81 with the brunch r83, without success.  Again, I turned off Google Sync._

In both cases, I ran the sudo resize-data command, to see if that would help.

I tried the dmesg but wasn't sure what I was suppose to be looking for.  

Any other suggestions that I can try, please let me know.
Thanks again for all your help!

sebanc, 2020-06-07:
I am not sure, you correctly performed this:

> Could you try deleting google sync ?
https://support.google.com/chromebook/answer/2914794?hl=en

The point is not to disable sync but to remove synced data from your google account.

Also, what's your laptop model ?

dshnrob, 2020-06-07:
Okay, I did as asked, and removed the synced data.  I rebooted and it still unable to build the Google Play store.

The model I am using is an HP Touchsmart 520.  Although a PC, it is built on a Laptop chassis.

And I am booting from a 32GB USB drive.  Is there anything else you can suggest?

Thanks again for all your help!

sebanc, 2020-06-07:
Could you attach a dmesg to this issue ? (run "dmesg > ~/Downloads/dmesg.txt" and attach dmesg.txt from your downloads folder to this issue)

dshnrob, 2020-06-07:
Hi sebanc,
Here's the attached dmesg.txt file:
[dmesg.txt](https://github.com/sebanc/brunch/files/4742748/dmesg.txt)

sebanc, 2020-06-08:
In theory it should work, I am not sure what's the issue here.

1) Verify in your bios that graphics / virtualisation options are enabled and set to the max.

2) The only notable error I can see in the dmesg should be solved by adding "nohugeiomap" to the kernel command line. Could you press "e" when you see grub, add "nohugeiomap" after "cros_debug", boot with CTRL+X (or F10) and see if android apps work ?

3) If it did not work, I would need you to attach the /var/log/arc.log file.

dshnrob, 2020-06-08:
virtualization : enabled
graphics : maxed
nohugeiomap : added

Unfortunately, still no Google Play Store.  Was wondering if I needed to enable Hyper-Threading.

Here's the arc.log.

Thanks again for all your help.
[arc.log](https://github.com/sebanc/brunch/files/4747210/arc.log)

sebanc, 2020-06-08:
Yes, enable hyperthreading it could actually be related. I will have a look at the log

dshnrob, 2020-06-09:
Enabled hyperthreading but Google Play Store still not working.

Anything else I can try, please let me know.
Thanks

sebanc, 2020-06-29:
Hi,

Could you try the latest release here ? (validate the github email invite first)
https://github.com/sebanc/brunch-testing/releases

darethehair, 2020-07-01:
Not sure if my experience will be helpful, but I also experienced the Playstore app never completing a load on my Acer C720 running Rammus recovery.  Just now I reloaded my USB stick with 'brunch_r83_k4.19_unstable_20200630.tar.gz' and re-tried.  This is what happened:

- Playstore did the rotate-but-never-complete-load as before
- Everything froze for about a minute
- Playstore app actually loaded (!)
- Search bar would not allow me to type anything
- Screen went black
- Original desktop appeared
- Tried to start Playstore app again, but just rotate-but-never-complete-load

Conclusion: perhaps a bit more progress than previous versions of Brunch, but (for me) not success.

sebanc, 2020-07-01:
I post it a lot so you have probably already done it but in case you have not done it you can try deleting google sync ?
https://support.google.com/chromebook/answer/2914794?hl=en

darethehair, 2020-07-01:
> I post it a lot so you have probably already done it but in case you have not done it you can try deleting google sync ?
> https://support.google.com/chromebook/answer/2914794?hl=en

I used a Google account that doesn't really have anything in it, but I did go ahead of do a 'reset sync' as mentioned (the menu path to this for me is slightly different than mentioned).  Unfortunately, Playstore just rotates forever and never loads :(

sebanc, 2020-07-01:
Could you try modifying /etc/init/udev-trigger.conf by replacing:
`start on starting failsafe`
with
`start on stopped udev-trigger-early` ?

sebanc, 2020-07-01:
Just to clarify, it is a shot in the dark as I don't really understand those playstore issues but maybe worth testing.

darethehair, 2020-07-01:
> Could you try modifying /etc/init/udev-trigger.conf by replacing:
> `start on starting failsafe`
> with
> `start on stopped udev-trigger-early` ?

**This appears to have worked!**  I didn't know what to do (if anything) to 'activate' this change, so all I did was to re-try starting the Playstore app -- and it opened immediately with no delay!  I then easily installed one of my daily apps, and it works fine, from what I can tell :)

So, it would seem that whatever 'magic' you had in that unstable 'brunch_r83_k4.19_unstable_20200630.tar.gz' package, combined with the '/etc/init/udev-trigger.conf' change you suggested, may be the secret-sauce that is needed for Playstore to work on computers like my Acer C720 (I had no problems with any of this stuff on my ASUS and Dell Chromebooks, so I don't know what the problems are/were).

Of course it would be good if others could verify all of this.

sebanc, 2020-07-02:
Thanks for the feedback :)

I included this patch in the new unstable build here:
https://github.com/sebanc/brunch-testing/releases

Could you try it ?

daemongloom, 2020-07-02:
> Thanks for the feedback :)
> 
> I included this patch in the new unstable build here:
> https://github.com/sebanc/brunch-testing/releases
> 
> Could you try it ?

That version breaks linux(beta) for me.
`Error: routine at frontends/vmc.rs:151 `vm_start(vm_name,user_id_hash,features)` failed: bad VM status: `VM_STATUS_FAILURE`: Failed to mount shared directory`

sebanc, 2020-07-02:
@daemongloom Is it the standard crostini (works for me) or is it the arch crostini vm ?

Regarding the other point, I am not sure why you want to disable the 10s device sleep as your device needs it as all others (maybe you only need 1s delay to boot from hdd, 3s to boot from sdcard, 4s from usb... I have no idea for your hardware). Brunch purpose is to be compatible with most systems but feel free to unpack the initramfs, change the sleep delay and repack it.

daemongloom, 2020-07-02:
> @daemongloom Is it the standard crostini (works for me) or is it the arch crostini vm ?
Standard crostini.

> Regarding the other point, I am not sure why you want to disable the 10s device sleep as your device needs it as all others (maybe you only need 1s delay to boot from hdd, 3s to boot from sdcard, 4s [...]
That's why I'm asking if it is possible to add option like no_ui_delay. It's not a problem for me to build it from source, but there are no public sources for your unstable branch. Also, other people would benefit from it greatly, as most internal ssd's require no additional sleep.

sebanc, 2020-07-02:
> Standard crostini.

Ok I have another report, I am looking at it. If you can backup your current container, I would be interested to know if you can setup a new crostini container under the new release?

> That's why I'm asking if it is possible to add option like no_ui_delay. It's not a problem for me to build it from source, but there are no public sources for your unstable branch. Also, other people would benefit from it greatly, as most internal ssd's require no additional sleep.

I make too much changes currently, testing new things that I cannot maintain a completely up-to-date sources but I expect to be able to improve this soon.
Regarding the delay, on the same hardware, a newer kernel version might take longer to initialise a block device than the previous kernel version (happened to me while creating brunch). If I remove this delay (or give the possibility to do so), I will be crawling under chromeos not booting issues in no time.
Unpacking and repacking the initramfs can easily be automatised and is still an option.

daemongloom, 2020-07-02:
> > 
> 
> Ok I have another report, I am looking at it. If you can backup your current container, I would be interested to know if you can setup a new crostini container under the new release?

"vmc start termina" already causes error, so it dies even before starting container. I'll backup my own container and test later in the evening.

darethehair, 2020-07-02:
> Thanks for the feedback :)
> 
> I included this patch in the new unstable build here:
> https://github.com/sebanc/brunch-testing/releases
> 
> Could you try it ?

The newest version of Brunch seems to work well with Playstore right out-of-the-box (though a small glitch occurred the first time I tried to view 'my' library).  This is nice!  However, I did go on to try installing Linux Beta because of what daemongloom reported, and I got the message 'Error downloading the virtual machine'.  However, I had not yet ever tried doing this on my Acer C720, so I am not sure if something 'broke' or if it ever was capable of running Linux (?).

daemongloom, 2020-07-02:
> > 
> 
> Ok I have another report, I am looking at it. If you can backup your current container, I would be interested to know if you can setup a new crostini container under the new release?

No, setting up a new container is not possible. It shows a error in the end. Rolled back and restored backup - everything works again.

sebanc, 2020-07-02:
@daemongloom I think I am starting to understand the issue and it seems related to updating an existing chromeos. If you have the time, could you confirm whether the default crostini works on a fresh install and also if you are able to restore your crostini backup ?

daemongloom, 2020-07-03:
> also if you are able to restore your crostini backup ?

Restoring backup on upgraded installation also fails. Will try today to use it on USB.

sebanc, 2020-07-03:
Just posted a new unstable build on which crostini should be working after an update.

daemongloom, 2020-07-03:
> Just posted a new unstable build on which crostini should be working after an update.

Linux apps work fine now. Is clean installation test  still required?

sebanc, 2020-07-03:
@daemongloom no, I think I understand the whole thing now. Thanks for the help and the feedback :)

sebanc, 2020-07-05:
The patch discussed here has been included in the latest release but appears to be breaking android apps for some...

@darethehair Could you install the latest release on your C720, add "options=original_init" to the kernel command line and try replacing the below text in /etc/init/failsafe.conf:
`start on starting system-services or stopped failsafe-delay`
with
`start on stopped udev-trigger-early`

Do android apps work after this ?

closed 2026-04-27 by github-actions[bot]