sebanc/brunch #273: chrome os was not found on any device, fallback to shell ?

Herzog1701, 2020-06-06:
Hi,
during boot up from the Chrome OS Stick on my old **HP Compac nx7400** and also my old **AsusEee PC Seashell series Laptop** it stops, showing:
"Chromeos was not found on any device, fallback to shell?"

I don't know why this happens, because on other legacy machines Chrome OS works flawless.
Can anybody help?

sebanc, 2020-06-06:
Hi,

This happens because "core 2 duo" processors and "atom 1st gen" do not support the cpu instructions included in current chromeos recovery images.

The built from source chromiumos distros (cloudready, fydeos...) might still work

Herzog1701, 2020-06-06:
Hi sebanc,
thanks for your reply.
Good to know! this saves me time.

thx
Mike

abdullahemad12, 2020-06-16:
did it work ?

Herzog1701, 2020-06-16:
Hi it worked on x5-Z8350 but not on Atom Gen1.
thx

closed 2020-06-26 by sebanc