sebanc/brunch #274: Wi-Fi BRCM [ BroadCom ] drivers

lMarenda, 2020-06-06:
are BRCM drivers already supported ? 
we're using Brunch R81 and Chrome OS Rammus 81 however  BT switches on only and no devices were ever found though, Wi-Fi network never turns on then we can't login. 
Broadcom Chipset:	20702B0 ( BT ) 
Broadcom BCM43xx 1.0 (7.21.190.32 AirPortDriverBrcm4360-1315.1) 
thanks' for your effort we really appreciate the work you have done.

luisa

sebanc, 2020-06-07:
Hi,

Have you tried adding "options=broadcom_wl" to the kernel command line ? (run "sudo edit-grub-config" in crosh shell and add "options=broadcom_wl" after "cros_debug")

lMarenda, 2020-06-07:
will try soon. 
 unfortunately GRUB configuration file no longer contains  ChromeOS boot options in fact, we may suggest to also add a few  lines with text and examples of menu entries [ for GRUB config  ] https://github.com/sebanc/brunch#dual-boot-chromeos-from-your-hdd 

that would help US when  we may  just create an entry  manually  and options we need to  restart with ChromeOS image already located in a disk partition. 

<blockquote>Hi,

Have you tried adding "options=broadcom_wl" to the kernel command line ? (run "sudo edit-grub-config" in crosh shell and add "options=broadcom_wl" after "cros_debug")</blockquote>

sebanc, 2020-06-07:
Here is my script if you need it, i directly added "options=broadcom_wl":
```
menuentry "ChromeOS" {
	img_part=/dev/nvme0n1p8
	img_path=/chromeos.img
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
		cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path options=broadcom_wl
	initrd (loop,gpt7)/initramfs.img
}
```

lMarenda, 2020-06-07:
nice !

lMarenda, 2020-06-07:
very nicely done. 
Awesome project  and we-re so impressed we can even resize the screen to adjust unwelcome overscan issues, when connecting to HDMI external display. 
we know we don't like ETA questions  however, is there a  chance to test nVidia GPU  drivers ? 
we noticed,  during startup, some description with nVidia  hardware show up .

sebanc, 2020-06-08:
Wrt nvidia, It depends on what you mean:

- If you mean a dual gpu setup with both intel and nvidia, I left the nvidia driver enabled in the kernel so that certain functions like hdmi out can be used. However, there is no userspace driver for nvidia graphics in chromeos but I don't think that chromeos with its current design would really benefit from using a dGPU. Also, if needed to play games, the dGPU can apparently be activated in crouton.

- If you mean nvidia as single gpu, the chromeos android apps framework is currently not compatible with nvidia gpu, cloudready therefore already provides a similar experience.

lMarenda, 2020-06-08:
HI and thanks'  once more for your very clear message replies to our earlier ones. 
 
we kind of thought...  if you can make use of nVidia GPU fitted on an old macbook and also running a linux distro,  though making no use of nVidia drivers/modules and using alternative ones instead, we can eventually exploit the very same customized kernel [ and maybe appropriate kernel modules ] while starting up a ChromeOS " instance ", on the same hardware platform [ we had no clues about user-space for nVidia GPU  and about limitations of android framework  ] 
   
we're in the process of  migrating everything away from apple mac OS however touchpad linux drivers aren't working  well,  those drivers [ maybe kernel modules ]  aren't anything close to  original apple kext files and it really feels awkward when using an old macbook touchpad combined with LinuxMint OS [ for instance. ] 
linux distro won't make  the transitioning away from apple OS [ and its ecosystem ]  anywhere close to an easy transition. 

off-topic  note: 
[ we have this on-going project to make assessments on how easy can be to switch to different and at times better OS environment, when using old PC/laptops on a daily basis however, we don't want to be forcefully stuck with linux distro are, at times, " freaking out " most non-power users in EU ] 

WE'LL MOVE THESE LAST MESSAGES TO A NEW THREAD WHILE ALSO CLOSING BCM [ BROADCOM ] WI-FI CARD  ISSUES.

closed 2020-06-08 by lMarenda