sebanc/brunch #276: Lenovo d330 convertible camera orientation

cino31, 2020-06-06:
Hello ,

Thank you again for this Nice project.

Everething work fine on my Lenovo D330-10igm convertible except the selfie camera which is rotated by -90 degrees angle ,how can I ajust the camera rotation ? 

I'm on Brunch v80 stable and rammus v80 for the moment and everething work fine except the camera orientation.

I tried last brunch and rammus also but graphics not work since you introduced i915 bug fix.

sebanc, 2020-06-07:
Hi,

I think that on this model it is actually the screen which is 90° off, could you rotate the screen in settings ?

sebanc, 2020-06-08:
I included a fix for the display totation in the new "brunch-testing" repository release, could you try it ?
https://github.com/sebanc/brunch-testing/releases

cino31, 2020-06-08:
Hello ,
I just installed the new brunch testing and now the front camera is absolutely fine & in the good orientation . Thank you for the fix.
But now i have graphics card problems ,i have to reboot several time to get display working and if use tablet mode and press power button the display go black but never go ON again ,

The probleme is also present in brunch v80 stable but when i press the power button twice ,the display work again without any problem ,but the button trick don't work on newer brunch even the test one that i just installed.

cino31, 2020-06-08:
The problem is also present with linux ,only in the intel N4000 CPU with HD Graphic 600 iGPU ,
There is many treads talking about this problem ,and I think a fix in present in the last linux karnel update.
just google "i330-10igm n4000 i915 bug"

In v80 brunch my screen work normally 8 Time on 10 ,but if I have a black screen on boot or after sleep: I just click Twice on power button and Boom magic the screen work again.
But on newer brunch the screen go black when I press power button but never go On again when I press power button again.

Ebbot9876, 2020-06-11:
My Chuwi Hi10 X also has 270° camera orientation!

sebanc, 2020-06-13:
@cino31 The fix you mention is already in the kernel, also I have another user with the exact same device and he does not have the screen not working 2 times out of 10 issue. Is your bios up-to-date ?
The suspend issue is confirmed (not only for this device) and I am working on it.

@Ebbot9876 I might be able to do something for your camera orientation issue but I will need you to post a dmesg (run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and attach dmesg.txt from your downloads folder)

closed 2020-06-26 by sebanc