sebanc/brunch #277: Huawei Matebook X Pro Issues

Vatri91, 2020-06-07:
Hello,

thank you very much for your work on the project. Finally, I was able to use chrome os for the first time. I am sending a list of things that are not working properly. 

Don't Work:
- Fingerprint Reader | Goodix GXFP5187 (No driver on linux)
- light sensor/No auto brightness (Works on Ubuntu)
- FN Keys: (Works on Ubuntu)
   * F7 - Mute Microphone 
   * F8 - Change Monitor
   * F9 - On/Off wifi
   * F10 - Open Settings
- Right Click on Touchpad (Works on Ubuntu)

Issues:
- Touchpad - I have currently disabled the "click by touching" function. An attempt to move the cursor is treated as a click. Using "alt_touchpad_config" doesn't help
- Audio in android apps - The sound in android applications is very quiet and distorted.
- After starting the laptop, the screen is very bright at the login screen. Every time after logging in I have to reduce the brightness. The brightness level settings are not saved
- Spotify (Linux) - I cant login in. I get a firewall problem message
- The system takes a long time to start

I hope you can improve this in updates

Best,
Adrian

Vatri91, 2020-06-19:
One update. When I open the video on YouTube using Chrome and at the same time launch the Android application, the sound works normally.

Any ideas on how to solve a sound problem permanently in Android Apps?

sebanc, 2020-06-20:
Hi,

Most of the points you mention are standard chromeos behavior:
- In chromeos, brightness restores to a hardcoded level on each boot
- The mapping for function keys is not always compatible (chromebooks have no hardware keys)
- ...

The only issues I see are:
- ambient light sensors, unless someone writes a driver for your sensor, it is not supported (#306).
- touchpad, you can refer to #305 and try to find a configuration which works for your device.
- sound, it is the first time I see an issue about sound working in chromeos but not correctly in android apps. The fact that sound works in chromeos shows that the driver is correctly included but that the android container does not handle it well. Unfortunately, I cannot modify the android container so the only idea I have would be to try changing some audio related chrome flags.

Vatri91, 2020-07-04:
Hi,

about sound. Yes sound in system, but when I boot the system, I see a message. 
0.800356] Alsa device list:
0.802107] No soundcards found.

Maybe this is a problem somewhere in the drivers

sebanc, 2020-07-04:
This part of the log is normal, it's just because sound drivers are not yet loaded at that point as they are built as modules.

Sorry but I still don't have any idea how to fix your issue. Maybe check in your bios if there is any "audio enhancement" parameter or something like that can be disabled.

steinano, 2020-07-18:
I fixed the touchpad issue by editing
sudo nano /etc/gesture/40-touchpad-cmt.conf 

and change from 1 to 0
 Option          "Tap Minimum Pressure" "0"

Vatri91, 2020-07-18:
> I fixed the touchpad issue by editing
> sudo nano /etc/gesture/40-touchpad-cmt.conf
> 
> and change from 1 to 0
> Option "Tap Minimum Pressure" "0"

<3

steinano, 2020-07-18:
I was too quick. The issue seemed resolved for a while, but now came back. On mine it's working normally for single taps, but every two finger scrool is initially registered as a right click. This didn't happen after I changed this setting (tried many values) for 20-30 minutes, but now seems to have reverted.

steinano, 2020-07-21:
After more troubleshooting the touchpad issue I found that sometimes it works normally as it should. Then after a few minutes it start misbehaving again (all two finger scrolls registers as clicks). 

This leads me to believe it's not a settings thing, could it be that some driver crash ?

Vatri91, 2020-07-31:
Another bug appeared after the beta update.

The sound in the entire system is extremely quiet. Previously, this issue only affected Android apps. The sound works normally on headphones (Chrome and Android).